microsoft/Azure-Kinect-Sensor-SDK #1591: Issues with Azure Kinect Body Track Code

ZhenyueQin, 2021-05-04:
Hi, 

I am a developer of Azure Kinect. I tried to build the code in the following link with Visual Studio 2019: 
https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/csharp_3d_viewer

However, I could not get the GPU working for estimating joints. I can see the red light is on for Azure Kinect, but nothing is showing up. 

For more details, please refer to this question that I asked on StackOverflow: 
https://stackoverflow.com/questions/67302411/visual-studio-nuget-restore-fails-for-microsoft-azure-kinect-bodytracking-depend

My GPU is NVIDIA 3060, and the OS is Windows 10. 

Using CPU seems to be fine, although the skeleton estimation is lagging and slow. 

Thank you very much!

qm13, 2021-05-06:
Does the pre-built k4abt_sample_3d_viewer.exe included in /tools work?
What GPU mode are you using?
What version of Windows 10 are you using?
What version of CUDA are you using?
Do you have the necessary CUDA / TensorRT DLLs in place?

ZhenyueQin, 2021-05-07:
pre-built k4abt_sample_3d_viewer.exe works fine. 
What is a GPU mode? 
CUDA 11.1.

closed 2021-08-16 by qm13