microsoft/Azure-Kinect-Sensor-SDK #1599: Turn off point cloud compensate feature

fi6, 2021-05-17:
**Describe the bug**
Points are filled in the empty area which is a disturbance in some high-accuracy applications.

**To Reproduce**

**Expected behavior**
Add a 'high accuracy' parameter or related ways to turn off the compensate feature.

**Logs**

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04
 - SDK Version:  1.4.1
 - Firmware version: rgb: 1.6.102, depth: 1.6.75

**Additional context**

closed 2021-08-16 by qm13

fi6, 2021-08-16:
Hi @qm13,

Why the issue is closed? Is there any comments/solution for this?

v-wewei, 2021-11-12:
@fi6 Does the issue solved?