microsoft/Azure-Kinect-Sensor-SDK #1600: Unity BodyTracking Sample catching exception for background thread result = K4A_RESULT_FAILED

naotoerg, 2021-05-18:
**Describe the bug**

unity Example stops catching exception for background thread result = K4A_RESULT_FAILED 

I have found possible typo on MoveLibraryFiles.bat.
`
copy %BODY_TRACKING_TOOLS_PATH%cudnn64_cnn_infer64_8.dll./
copy %BODY_TRACKING_TOOLS_PATH%cudnn64_ops_infer64_8.dll 
`
Azure Kinect Body Tracking SDK did not have such two files.
But it had 

cudnn_ops_infer64_8.dll
cudnn_cnn_infer64_8.dll

Unfortunately fixing these two possible typo didn't fix the K4A_RESULT_FAILED issue.

[image]

  
**To Reproduce**
Execute all the steps on the readme.
open sample scene on unity and play.

**Desktop (please complete the following information):**
 - OS with Version:19042.928 
 - SDK Version: Azure Kinect SDK v1.4.1 / Azure Kinect Body Tracking SDK v1.1.0 
 - Firmware version:  Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
-Unity Version: 2019.1.14f1

**Additional context**
k4aviewer and k4abt_simple_3d_viewer works flawlessly.

naotoerg, 2021-05-20:
I got the sample project working.
Not really a solution but, I copied all the dlls from the SDK to the unity project and now it works.
I assume that the batch file is missing some files to copy.

WhimsicalToyBox, 2021-05-20:
It's not a radical measure, but if you use ```TrackerProcessingMode.Cuda``` instead of ```TrackerProcessingMode.Gpu``` when initializing Tracker, it will work for the time being.

https://github.com/microsoft/Azure-Kinect-Samples/blob/master/body-tracking-samples/sample_unity_bodytracking/Assets/Scripts/SkeletalTrackingProvider.cs#L46

naotoerg, 2021-05-21:
> It's not a radical measure, but if you use `TrackerProcessingMode.Cuda` instead of `TrackerProcessingMode.Gpu` when initializing Tracker, it will work for the time being. [...]

Thank you for the tip!
Cuda mode works a lot faster then GPU mode!

UnaNancyOwen, 2021-06-15:
This problem will be solved with https://github.com/microsoft/Azure-Kinect-Samples/pull/62.

closed 2021-08-16 by qm13