microsoft/Azure-Kinect-Sensor-SDK #1605: I want to know how to get the Rotator of AzureKinect Device

VegetableWithChicken, 2021-05-21:
Yes, I want to get the rotation of the device itself. For example, when I pick up the camera and rotate it, I can get a rotation angle information of myself

saulthu, 2021-05-21:
Getting IMU samples and using the accelerometer vector gives you orientation relative to gravity, assuming the camera is not being moved. This doesn't give a heading direction however.

VegetableWithChicken, 2021-05-21:
> Getting IMU samples and using the accelerometer vector gives you orientation relative to gravity, assuming the camera is not being moved. This doesn't give a heading direction however.

so nice for u ,but i just want to know the relative rotator  to device itself, the data of rotator will help me to transform the other camera.

saulthu, 2021-05-21:
You want rotation to the device relative to what? Sounds like you want it relative to another camera. That is something you must work out yourself, it is not provided by the device i.e. they provide no extrinsic calibration parameters, nor any way built-in to find them.

saulthu, 2021-06-09:
The provided registration example is probably a good place to start: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/calibration_registration

closed 2021-10-06 by qm13