microsoft/Azure-Kinect-Sensor-SDK #1606: k4abt_sensor_orientation_t not working anymore

Moovency, 2021-05-21:
Hello,

The sensor_orientation parameter seems to not be taken in account anymore.
No matter the value we set, bodies are only detected with the camera beeing on its normal position.

Could you check if you have the same issue please ?

Thank you.

Moovency, 2021-06-03:
Hello,

Any update on the subject ?

Thank you.

scotthsu98052, 2021-06-05:
Please provide more info:
  BT SDK version
  OS version
  GPU model
  tracker_config.processing_mode
  Do you use default model file or you use other model files?

Moovency, 2021-06-09:
BT SDK: 1.1.0
OS version: Windows 10.0.19042
GPU model : Nvidia Geforce GTX 1050
tracker_config.processing_mode : K4ABT_TRACKER_PROCESSING_MODE_GPU
For model file we use dnn_model_2_0_lite_op11.onnx and dnn_model_2_0_op11.onnx
The error appears with both of them.

PierrePlantard, 2021-10-11:
Dear @qm13, is this issue is corrected ? If so, when do you plan to release the new version.
Regard,
Pierre

fkavache, 2021-12-27:
@Moovency 
Could you please provide more detailed information on how exactly are you reproducing this issue?
Also, it's interesting whether bodies are detected only partly, sometimes or they are not detected at all.

Moovency, 2022-01-03:
Hi @fkavache 
To reproduce this issue : 
- call **k4abt_tracker_create** with **k4abt_tracker_configuration_t config** and set **sensor_orientation** with **K4ABT_SENSOR_ORIENTATION_FLIP180**  (this is reproducible also with the other options https://microsoft.github.io/Azure-Kinect-Body-Tracking/release/1.1.x/group__btenums_ga8e5fb3391addee8fadaa809c601e223e.html ).
- Flip the Kinect Camera 180°
- Try to track the skeleton

-> In our case, the body is not tracked at all when using 180° (rotation is not taken in account)

Let me know if you need more details.

fkavache, 2022-01-10:
Hi @Moovency

In my case body tracking is working fine. 
In the worst scenario (flipped camera without any orientation parameter) at least there should be partial body recognition. 
Can you try to:
1. record flipped .mkv video using k4arecorder tool and use it as an input? 
(you can use the PlayFile function from [simple_3d_viewer)](https://github.com/microsoft/Azure-Kinect-Samples/blob/master/body-tracking-samples/simple_3d_viewer/main.cpp) sample as reference.
2. flip camera 180° without any orientation parameter.

Moovency, 2022-02-04:
Hi @fkavache 

You can check in attachment the scenarios we have tested : 

- Pic01 : k4abt_sensor_orientation_t set to K4ABT_SENSOR_ORIENTATION_DEFAULT , camera in place -> body tracking ok
- Pic02 : k4abt_sensor_orientation_t set to K4ABT_SENSOR_ORIENTATION_FLIP180, camera in place -> body tracking ok
- Pic03 : k4abt_sensor_orientation_t set to K4ABT_SENSOR_ORIENTATION_FLIP180, upside down camera -> body tracking ko

Thanks !

[image]

[image]

[image]

fkavache, 2022-02-18:
Hi @Moovency

I rechecked just in case, I can confirm that BT **works correctly** with FLIP180 option and camera upside down.

closed 2022-03-21 by qm13