microsoft/Azure-Kinect-Sensor-SDK #1607: request publish the build scripts for open-source parts of body tracking (e.g. onnxruntime)

diablodale, 2021-05-24:
I request the build scripts/instructions for all the open-source components used by the body tracking sdk.

For example...document how to build `onnxruntime.dll`, `onnxruntime_providers_shared.dll`, and `onnxruntime_providers_tensorrt.dll`. These are already [owned and open-sourced by Microsoft](https://github.com/microsoft/onnxruntime) itself. And if there are more such components, do the same for them.

### Is your feature request related to a problem? Please describe.

Issue https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1576#issuecomment-825862494 documents a memory leak which might be fixed in a newer onnxruntime. But since these DLLs are custom compiled, we can't test for this leak fix, can't upgrade the onnxruntime for future execution providers, performance/feature updates, etc.

### Describe the solution you'd like

I prefer to see a cmake CMakeLists.txt that simply and clearly builds these DLLs (and linux libs).
Alternatively, a markdown file which details the needed steps to build the DLLs and linux libs.

### Describe alternatives you've considered

None known. Without the specific build parameters/setup, it is not possible for a consumer of the SDK to update these components.

qm13, 2022-05-19:
@diablodale this is covered in the ONNX Runtime [repo ](https://github.com/microsoft/onnxruntime) README

closed 2022-05-25 by qm13

diablodale, 2022-05-25:
@qm13, that is both false and not the OP.

> Without the specific build parameters/setup, it is not possible for a consumer of the SDK to update these components.

There are 98 boolean compile switches + additional switches that control MSVC and Cmake.
If I take only the 98 compile switches for onnxruntime, there are 2^98 possibilities or 316912650057057350374175801344 possible choices for switches.

My OP continues to be, what are the specific build parameters to compile ONNX so it is compatible with this SDK.

diablodale, 2022-09-01:
Further, it is not possible for an application to reasonably use the ONNX runtime since K4ABT v1.1.2 errantly checks version, errantly skips `__pfnDliNotifyHook2` and bug #1510

The error reported directly above in TrackerHost.cpp (184) is using an unknown method to test for the DLL that is incompatible with delayloading/redirection of a DLLs with  `__pfnDliNotifyHook2`. The hook is not called and therefore your unknown version check method is preventing the hook redirecting K4ABT's errant DLL load to your custom DLL renamed to, e.g. `onnxruntime_kinect3.dll` so that it doesn't conflict with the main app.

This is another example of K4A/BT expecting to be a monolithic application where the kinect is the only device and kinect body tracking is the only neural net running with ONNX.