microsoft/Azure-Kinect-Sensor-SDK #1610: Support for ARM/ARM64

rjam-xav, 2021-05-26:
Hello,

Is the Microsoft Kinect SDK supported on ARM64 architectures like the Jetson Nano? The website still lists hardware requirements as x86 architecture but I see the git show support for ARM as well. 

Thank you.

taai, 2021-05-27:
@rjam-xav Yes, it works with *Jetson Nano*, I'm using it right now (Jetson Nano Developer Kit, 4GB).

On the SD card install the "Jetson Nano Developer Kit SD Card Image" as an operating system. https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit

Fallow these instructions to install Azure Kinect SDK (use instructions for "Ubuntu 18.04"): https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download and https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md and maybe https://github.com/valdivj/Azure-for-Kinect-Jetson-nano

Right now I can't write a full instruction about how to set this all up... There definitely are some things that differ from the instructions...
As I remember, instead of running this
```bash
sudo apt-add-repository https://packages.microsoft.com/ubuntu/18.04/prod
```
you must run this
```bash
echo "deb https://packages.microsoft.com/ubuntu/18.04/multiarch/prod bionic main" | sudo tee -a /etc/apt/sources.list
```

And if you are planning to use the ```k4arecorder``` without the displays (run from SSH or run as a service/daemon), you must set the ```DISPLAY=:0``` environment variable. Here is a full command I am using for my project:
```bash
DISPLAY=:0 k4arecorder output.mkv
```

By the way, *Raspberry Pi 4* is not supported (incompatible OpenGL versions or something like that) – I actually tried that first.

qm13, 2021-05-27:
Sensor SDK does support Jetson Nano range of ARM64 devices.

closed 2021-07-08 by qm13