microsoft/Azure-Kinect-Sensor-SDK #1614: Cannot Connect to Kinect with ROG Strix TRX40-E or NVIDIA 2080 Super USB-C

danzeeeman, 2021-06-01:
I have tried updating the motherboard's firmware and have tried the USB-C port on my NVIDIA 2080 Super and cannot get the device to show up.  I have performed a factory reset of the device with the reset button.  I have followed all troubleshooting and still cannot get the device to show up on my windows machine.

diablodale, 2021-06-03:
The kinect azure is a webcam. And more.
You should be able to plug the Kinect Azure into a USB-C port. And then you can use it in Skype, Zoom, Camera app, etc.
If it does not work in any of those apps, it is probably a hardware problem. You might have a defective Kinect.

danzeeeman, 2021-06-25:
it was a defective kinect

closed 2021-06-25 by danzeeeman