microsoft/Azure-Kinect-Sensor-SDK #1615: ..\src\TrackerHost\TrackerHost.cpp (185): Create(). Find onnxruntime.dll at ...\x64\Debug\onnxruntime.dll but it doesn't load correctly!

Fugashu, 2021-06-01:
**Describe the bug**
We switched from Nvidia 980Tis to Nvidia 3070 RTX GPUs, included the latest Sensor and Body Tracking SDK (without NuGet) in our Visual Studio C++ Project. The Sensor SDK works totally fine. Unfortunately the Body Tracking SDK does not anymore when trying to create a k4abt_tracker instance. All DLLs were copied to the VS projects.

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 Pro Build 19042.746
 - Sensor SDK Version: 1.4.1
 - Body Tracking SDK 1.1.0
 - Firmware version:  1.6.110079014

**Additional context**
The Azure Kinect Viewer and Body Tracking applications work without errors.

Code snippet which shows the issue:

`k4abt_tracker_t tracker = NULL;
	k4abt_tracker_configuration_t tracker_configuration = K4ABT_TRACKER_CONFIG_DEFAULT;
	if (K4A_FAILED(k4abt_tracker_create(&calibration, tracker_configuration, &tracker)))
	{
		std::cout << "Failed to create body tracker!" << std::endl;
		k4a_device_stop_cameras(device);
		k4a_device_close(device);
		return 1;
	}`

Here as a screenshot:
[image]

Thanks for your help in advance!

diablodale, 2021-06-03:
I recommend you provide more detail. Your issue is too vague.

* Does the pre-compiled body tracking demo work? Run it in `C:\Program Files\Azure Kinect Body Tracking SDK\tools\k4abt_simple_3d_viewer.exe`
* Your separate cuda toolkit is irrelevant. You must use the exact DLLs that are distributed with the Body Tracking SDK. Including those specific CUDA DLLs...not your separate CUDA install.
* All DLLs and support files needed are in the above `tools` and `C:\Program Files\Azure Kinect Body Tracking SDK\sdk\windows-desktop\amd64`
* Turn on Trace logs for k4abt https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#collecting-logs
* Does this issue reproduce on other computers? Other graphic cards? Do body tracking work when you use CPU mode? etc.

Fugashu, 2021-06-03:
Thanks for your response.

- Yes, the pre-compiled demo works as expected without any issues.
- I separately installed the CUDA Toolkit, since I couldn't find the `cudart64_110.dll` inside `C:\Program Files\Azure Kinect Body Tracking SDK\sdk\windows-desktop\amd64` directory.  Apparently the `cudart64_110.dll` is inside `C:\Program Files\Azure Kinect Body Tracking SDK\tools`.

I replaced the dlls in my Visual Studio project with the dlls provided inside the `tools`-directory instead of the `sdk`-directory and it worked.

Thanks for your assistance!

closed 2021-06-03 by Fugashu