microsoft/Azure-Kinect-Sensor-SDK #1618: Kinect keeps failing

max246, 2021-06-04:
Using the latest SDK and latest firmware, during the usage with Unity, the kinect fails and disconnect from the PC and it shows up:
- Azura LED
- Azure kinect 4k  camera
but the depth sensor is not showing up.

Unplugging the cable, and plugging back didnt work and not even restarting the pc.

I managed to reinstall sdk and shown up again but then after an hour it failed again.

max246, 2021-06-04:
Picture of the list of devies [img](https://ibb.co/frc8SWq)

qm13, 2021-06-08:
What operating system are you using? What SDK and firmware?

max246, 2021-06-09:
> What operating system are you using? What SDK and firmware?

Using windows 10 with latest updates
Latest SDK 1.4.1

I think the issue might be connected to the motherboard using AsMedia usb controller, I am going to purchase a USB 3.0 PCI express which is using NEC chipset to confirm if that is the issue.
I read few people  having the same issue.

closed 2021-08-16 by qm13

max246, 2021-08-16:
Why closing this? It still happening and even got an azure ticket open for a while. 

It is just unclear why you cant provide proper hardware or solution to narrow down why of this issue.

qm13, 2021-08-16:
Duplicate of #1409

qm13, 2021-08-16:
@max246 did you try the Azure Kinect on a supported USB controller?

max246, 2021-08-17:
> @max246 did you try the Azure Kinect on a supported USB controller?

I have tried a new PC without asmedia usb controller but still happening, azure support is not really giving me a motherboard that is supported for "unknown reason" 
At the moment it seems like it is holding using the USB C from the rtx2070 but this is not really a solution , I mean it is but hardware specs such as supported motherboard should be provided if this is a usb issue.