microsoft/Azure-Kinect-Sensor-SDK #1620: Azure Kinect Camera freezes on its first capture

haeginh, 2021-06-08:
It is just as explained in the title. The Azure Kinect Camera just freezes right after showing very first frame.
The log was pasted below.
I have tried reset, firmware upgrade, connecting to the all the usb 3.0 ports I got.
The cpu of my laptop is intel i7 10th gen, and the graphic card is RTX2060.
OS is Ubuntu 18.04.

In addition, to be sure, I also tried with another Azure Kinect device (I have two of these).
However, I found the other one works so well.. which means this may be the problem of the device itself.

I have used this only for about <2 years, and the device was always on my desk... I used it only sometimes for about few hours in total.
It would be very appreciated if you could advise me what to do to make it work properly.

[ critical ] : ******************** Device Info ********************
[ critical ] : K4A SDK version:     1.4.1
[ critical ] : Serial Number:       000404694512
[ critical ] : RGB Sensor Version:  1.6.110
[ critical ] : Depth Sensor Version:1.6.79
[ critical ] : Mic Array Version:   1.6.14
[ critical ] : Sensor Config:       6109.7
[ critical ] : Build type:          Release
[ critical ] : Signature type:      MSFT
[ critical ] : ****************************************************
[ critical ] : ******************** Device Info ********************
[ critical ] : K4A SDK version:     1.4.1
[ critical ] : Serial Number:       000404694512
[ critical ] : RGB Sensor Version:  1.6.110
[ critical ] : Depth Sensor Version:1.6.79
[ critical ] : Mic Array Version:   1.6.14
[ critical ] : Sensor Config:       6109.7
[ critical ] : Build type:          Release
[ critical ] : Signature type:      MSFT
[ critical ] : ****************************************************
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    533544 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    566855 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    600200 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    633522 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    666855 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    700188 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    733533 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    766866 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    800200 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    833522 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    866855 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    900200 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    933533 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    966855 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1000188 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1033522 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1066866 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1100188 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1133533 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1166855 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1200200 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1233522 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1266866 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1300188 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1333533 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1366866 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1400200 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1433522 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1466855 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1500188 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1533533 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1566855 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1600188 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1633522 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1666866 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1700188 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1733533 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1766855 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1800188 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1833522 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1866855 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1900188 type:Color
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1933522 type:Color
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1966866 type:Color
[ error    ] : cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[ error    ] : depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[ error    ] : cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[ error    ] : depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063

haeginh, 2021-06-09:
I have tested several cases to clear out when it works and when it doesn't.
1. only color images -> works for all fps
2. color + depth (except for passive IR) -> works only for 15 and 5 fps (30 fps causes the problem specified above)
3. color + passive IR -> works for all fps
+ regardless of the fault in cameras, the other sensors keep works.

closed 2021-10-06 by qm13

arkader, 2021-11-02:
Hi @haeginh 
I've you managed to fix your issue. I'm using 6 sensors, and I have exactly the same problem.

eidetic-av, 2022-04-08:
For anyone still coming across this issue, it could be related to power. This was happening to me from a USB port that wasn't supplying enough amps. I changed to the USB port on the motherboard and was able to get color & depth images at 30fps without this error.