microsoft/Azure-Kinect-Sensor-SDK #1623: get_capture failed

LiuYi000, 2021-06-10:
[](url)**Describe the bug**
A multithread was used. One is for acquiring capture and one is for other image processing. while running for sometime(sometime a few minutes and sometime even more than 24 hours), sdk get_capture() return error and the device stopped working. 

 
**To Reproduce**
only happen on some specific PC, not easy to reproduce. 

 
**Expected behavior**
The camera can get capture well all the time even running for days.

[**Logs**]
[Uploading Log.log…]()

**Desktop**
Windows 10(1809) 64 bit
Intel(R) Xeon(R) Gold 6230 CPU @2.10GHz 2.10GHz(2 processors)
RAM 64.0GB (6.27GB usable)

**Software**
Visual studio 2019 C/C++
Camera SDK：1.4.0
Firmware version for RGB is 1.6.102
Firmware version for Depth is 1.6.75
color_format = K4A_IMAGE_FORMAT_COLOR_YUY2;
color_resolution = K4A_COLOR_RESOLUTION_720P
depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED
synchronized_images_only = true

_The code we get capture from camera shown as bellow:_
errCode = get_capture(&capture, std::chrono::milliseconds{ 1000 })
the error of this issue is errCode is false sometimes while using camera.

**Footnote**
1.	We used a 25m USB cable to instead of original USB line from camera as it is too short to satisfy our requirement.
2.	An error happens at 2021-5-27 19:00:29, the GPU usage at that time shown as following:
[image]

qm13, 2021-06-10:
Please upgrade to Sensor SDK 1.4.1 and latest firmware.

LiuYi000, 2021-06-11:
> Please upgrade to Sensor SDK 1.4.1 and latest firmware.

is this a bug in the current version I used and have solve in the latest version?

LiuYi000, 2021-06-16:
I haved upgraded to Sensor SDK 1.4.1 and latest firmware, but the issue still occurs.
this is the log for new version:

closed 2021-08-16 by qm13