microsoft/Azure-Kinect-Sensor-SDK #1624: Linux cant load depth camera

Ycao26, 2021-06-10:
**Describe the bug**
The depth sensor can be loaded when running on windows. But I run it on my Linux Ubuntu 18.04. It always shows same error messages:  Failed to start device! If you unplugged the device, you must close and reopen the device.

**To Reproduce**
opened the Azure Kinect Viewer through command  k4aviewer
click on 'open device' and 'start' with default options
error message pops up: Failed to start device! If you unplugged the device, you must close and reopen the device.

**Logs**
[ warning  ] : is_fw_version_compatable(). Firmware version for RGB is 1.6.98. Consider upgrading to 1.6.102 or newer. [ warning  ] : is_fw_version_compatable(). Firmware version for Depth is 1.6.70. Consider upgrading to 1.6.75 or newer. [ error    ] : depth_engine_start_helper(). Depth engine create and initialize failed with error code: 207. [ error    ] : depth_engine_start_helper(). OpenGL 4.4 context creation failed. You could try updating your graphics drivers. [ error    ] : deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper() [ error    ] : depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread() [ warning  ] : capturesync_add_capture(). Capture Error Detected, Depth  [ error    ] : dewrapper_start(). Depth Engine thread failed to start [ error    ] : dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start() [ error    ] : cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming() [ error    ] : depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063 [ error    ] : depth_start(device->depth, config) returned failure in k4a_device_start_cameras()

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: <Linux Ubuntu 18.04 >
 - SDK Version:  < 1.3.0 -->
 - Firmware version:  

**Additional context**

qm13, 2021-06-10:
Please try Sensor SDK 1.4.1

asergaz, 2021-06-11:
Same question on Microsoft Q&A: https://docs.microsoft.com/en-us/answers/questions/430571/cant-load-depth-camera-when-running-in-linux-ubunt.html?childToView=432249#answer-432249

Ycao26, 2021-06-13:
[image]
I meet an error when installing the 1.4.1 when using the command sudo apt install k4a-tools. My CPU is ryzen 9 3900x

Ycao26, 2021-06-13:
Could someone give me the whole command code of the installation of the SDK 1.4.1? Also, if I wanna copy 'scripts/99-k4a.rules' into '/etc/udev/rules.d/', should I just use sudo cp -a scripts/99-k4a.rules /etc/udev/rules.d/

asergaz, 2021-06-21:
@qm13 fyi can you follow-up. thanks

closed 2021-08-16 by qm13

ehdwo0427, 2022-10-06:
@Ycao26 The same problem has occurred. I can't run the camera in depth.

We used Ubuntu 18.04.