microsoft/Azure-Kinect-Sensor-SDK #1625: Body tracking crash on launch using 1.1.0 but works with previous version

benbenn69, 2021-06-11:
Hello everybody,

I meet a crash on launch of Body Tracking app using the last version 1.1.0. I'm using RTX 2060 and Azure Kinect
I tried with the previous one (1.0.1) and the app lauch automatically.
I filmed in slowmotion to have time to see what's the error (sorry for the quality)

[image]

Does anyone knows where the problem comes from ? 

Thanks so much,

Benb

diablodale, 2021-06-12:
You have not provide enough information to make much progress assisting. I recommend you follow the instructions on how to open an issue here and to include your detailed setup, your sample code, the reproduction steps to do to reproduce the problem, etc.

Other than than, the problem is listed on the first line. It found onnxruntime.dll but it doesn't load correctly. This is probably a setup problem. You will need to diagnose your setup. I recommend you read the documentation on how to setup and the troubleshooting documentation in the SDK.
In addition I recommend you try to run the example `k4abt_simple_3d_viewer.exe` in your K4ABT SDK tools directory. If that runs successfully, it is further evidence you have a setup problem with your app.

black4clover, 2021-08-15:
I solved this problem by downloading Microsoft Visual C++ Redistributable for Visual Studio 2015, 2017 and 2019.

https://support.microsoft.com/en-us/topic/the-latest-supported-visual-c-downloads-2647da03-1eea-4433-9aff-95f26a218cc0

closed 2021-08-16 by qm13

jjkislele, 2021-11-11:
> I solved this problem by downloading Microsoft Visual C++ Redistributable for Visual Studio 2015, 2017 and 2019.
> 
> https://support.microsoft.com/en-us/topic/the-latest-supported-visual-c-downloads-2647da03-1eea-4433-9aff-95f26a218cc0

It works. Thank you very much.