microsoft/Azure-Kinect-Sensor-SDK #1628: It‘s so  weird ,there is  high gpu  consumption (more than 35% ) even your shut down the tracker and just use cpu ,How to Fix this?

senshin998, 2021-06-21:
windows 10 ,
Kinect.Sensor version="1.4.1"
nvidia  gtx 1080ti 11G.

code:

    tracker_config.processing_mode = K4ABT_TRACKER_PROCESSING_MODE_CPU;// just use cpu
    k4abt_tracker_shutdown(tracker);//shut dowm 
  
result :
 gpu more than 35%

qm13, 2021-06-22:
Are you sure the GPU is being consumed by the body tracker? Windows 10 also uses the GPU.

senshin998, 2021-06-23:
Thanks for your reply,the GPU is not being consumed by the body tracker,the body tracker is being shut down , the GPU is consumed  by   the Kinect.Sensor  framework  itself ,
 When you shut down the tracker but do not stop the depth camera, the GPU will be consumed a lot, 
    
  VERIFY(k4a_device_open(cameID[cameIndex], &devices[cameIndex]), "Open K4A devices[cameIndex] failed!");
    k4a_device_configuration_t device_configs[camerasNum] = { K4A_DEVICE_CONFIG_INIT_DISABLE_ALL,K4A_DEVICE_CONFIG_INIT_DISABLE_ALL,K4A_DEVICE_CONFIG_INIT_DISABLE_ALL };
    device_configs[cameIndex].camera_fps = K4A_FRAMES_PER_SECOND_30;
    device_configs[cameIndex].color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32; 
    device_configs[cameIndex].color_resolution = K4A_COLOR_RESOLUTION_1440P;
    device_configs[cameIndex].wired_sync_mode = K4A_WIRED_SYNC_MODE_STANDALONE;
    VERIFY(k4a_device_start_cameras(devices[cameIndex], &device_configs[cameIndex]), "Start K4A cameras failed!");
     above  codes even does not create a tracker ,but still consumed   GPU more than 35%,  the Gpu reback to  normal(0%) if exec k4a_device_stop_cameras(devices[cameIndex]);
I'm pretty sure about that  and  this make me  confused,Why does this framework consumed GPU so much just for  running  camera?

senshin998, 2021-06-23:
[image]

diablodale, 2021-06-24:
First, task manager and all "usage graphs" are approximate and not absolutely true.

In my experience, there is a basic level of GPU use for the basic kinect azure functions. The system requirements https://docs.microsoft.com/en-us/azure/kinect-dk/system-requirements#minimum-host-pc-hardware-requirements suggest that a DirectX11 gpu is used for the "DepthEngine". The depthengine is the closed-source proprietary code that processes the raw sensor data->depthmap.

I also have seen that an integrated Intel GPU can *sometimes* show less relative "usage" during basic sensor use in task manager compared to a discrete Nvidia GPU. I believe part of this is due to the Intel using shared memory -- no memory transfers over PCIe both to/from the discrete gpu.

Other data, like color, may also be processed by the GPU if you use specific options. That part of the code is open-source and you can read the code yourself in this git repo. Intel gpus are much faster at decoding color compared to Nvidia. This is due to Intel's Quicksync decoder which in some configurations may be automatically used by Windows Media Foundation when you request color processing like BGRA32.

On my i7-10875h, integrated intel uhd graphics, discrete nvidia rtx2070super
with 640x576 depth and 1080p color. displaying onscreen both the depth and color images...

* forced intel uhd graphics is ~22% intel gpu, 0% nvidia
* forced nvidia is ~6% intel gpu, 12% nvidia

Your 1080ti graphics are 4 years old. It is capable of being used for the Kinect. However, it will likely have less performance than my graphics which were released ~6 months ago. In my perspective, I don't see anything alarming with your 31% usage that would be everything you do with the Kinect + Windows rendering itself.

senshin998, 2021-06-25:
Really thanks for your reply！ I‘m agree with task manager are  not absolutely true. but the azure kinect framework does consumed cudas in my case  which body tracker is shut down  and set K4A_COLOR_RESOLUTION_OFF , due to this fact,There are no enough cuda resources for running other neural networks and case  a lower fps ,  second,  I thiink raw sensor data->depthmap should be processesed on the CPU in real time just like other depth cameras such as intel  realsense ,  third，  gtx1080ti 11g graphics   is on par with   nvidia rtx2070super，the weirdest thing for last is  tha a  integrated intel graphics can fully meet It's work needs  but the It Occupies most of the resources when there is a powful graphics ,

senshin998, 2021-06-25:
Is there any way to assign device ID for  the  basic level of GPU use of    kinect azure functions ?   not  by setting  "tracker_config.gpu_device_id" ,It only for  body tracker  .

diablodale, 2021-06-25:
I do not know of an API today to force the depthengine to use a specific GPU. I do not think the SDK is currently designed to use a only a CPU. This has already been requested in https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1428

You can see the open-source api that it used between the open-source part of the SDK when it calls into the closed-source https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/include/k4ainternal/k4aplugin.h. I see no capability to chose a GPU.

I use separate gpu tools to force my app to run on a specific GPU during my testing. Window 10 now has a tool to choose an app's gpu in the advance graphic settings. Older Windows used the Nvidia control panel to specify the gpu for an app. This is the only way that I know to control gpu for the basic sensor data. The current SDK always needs a DirectX11 GPU. This is specifically stated in the system requirements I linked above.

I recommend you code and test your app. Until you do the work, you are guessing. I do not see any promise from Microsoft that you can process Kinect data locally + run additional neural networks. Instead, the docs focus on only the processing of the Kinect data. Microsoft will gladly provide Azure cloud services that can receive the processed Kinect data and run your neural nets. 😉

I often look at the task manager graphs when I am coding/testing. Recently, I was surprised. Task manager showed *less* Nvidia gpu use when I enabled features A+B --compared-- to only feature A. It seems impossible this could be true. But the task manager graph is not always accurate. Feature A requires very little gpu processing. I suspect more time is used copying the 4K data to+from the bus. My guess is that inaccurate task manager + the Nvidia gpu driver is batching/pipeline/parallel together transfers+work *and* the Nvidia hardware will change/scale its threading and clockspeed based on incoming work. All of these are probably making it appear there is less gpu use when I enable two features. This article https://devblogs.microsoft.com/directx/gpus-in-the-task-manager/ from Microsoft writes the % utilization you are looking at is "utilization of the busiest engine as a representative of the overall GPU usage". And a gpu has many engines (3d, compute, copy1, copy2, decode, encode, crypto, etc.).

senshin998, 2021-06-25:
thanks for sharing ，you help me a lot,    this code "_putenv("CUDA_VISIBLE_DEVICES=1")" may be partial  works，It force the tracker to use the specific GPU but  not valid ineffective in depthengine .

diablodale, 2021-06-25:
The body tracking has a good API to control the cpu, specific gpu. https://microsoft.github.io/Azure-Kinect-Body-Tracking/release/1.x.x/structk4abt__tracker__configuration__t_a663351b13a0caa2b0240abcbcbcd7b14.html#a663351b13a0caa2b0240abcbcbcd7b14

On the body tracking `CUDA_VISIBLE_DEVICES` would work if you also chose `K4ABT_TRACKER_PROCESSING_MODE_GPU_CUDA` using the body tracking api.

The doc I read https://docs.nvidia.com/cuda/cuda-c-programming-guide/index.html#env-vars suggests `CUDA_VISIBLE_DEVICES` is for CUDA api applications. The basic sensor sdk does not use CUDA api, therefore this variable probably does not work for basic sensor data. Do you have other doc or experience?

There might be a DirectML env var that works in body tracking. Last year the tensorflow-directml team added `DML_VISIBLE_DEVICES` https://github.com/microsoft/tensorflow-directml/releases/tag/v1.15.3.dev200911 and doc at https://docs.microsoft.com/en-us/windows/win32/direct3d12/gpu-faq#how-do-i-use-the--dml-visible-devices--environment-variable-to-control-which-gpu-s--get-used-by-directml. If you set `K4ABT_TRACKER_PROCESSING_MODE_GPU_DIRECTML` and this env var, you can probably control the gpu. The existing body tracking api above can do it also.

senshin998, 2021-06-26:
@diablodale thanks for your help ,you have good opinions, The basic sensor sdk does not use CUDA, funny thing is the basic sensor  occupies more gpu resources even  than  body tracking itself when you hanve powful card ,This caused me a lot of trouble，may you share the name of separate gpu tools?

diablodale, 2021-06-26:
> I use separate gpu tools to force my app to run on a specific GPU during my testing. Window 10 now has a tool to choose an app's gpu in the advance graphic settings. Older Windows used the Nvidia control panel to specify the gpu for an app.

Search in Windows 10 for "Graphics settings". When your language is not English, then look in Display settings. Then scroll down and look for graphics settings. It will take you to settings for "Graphics performance perference". You can choose your app and your gpu with that.

Before that was included in Windows 10, Nvidia control panel had the same setting. It is installed with Nvidia drivers. Run the Nvidia control panel and click on the Program Settings tab. Then you can choose your app, and choose your GPU.

closed 2021-10-06 by qm13