microsoft/Azure-Kinect-Sensor-SDK #1630: Cmake on transformation example cannot locate libjpeg-turbo even installed

xXCXMXx, 2021-06-28:
Consolidate compiler generated dependencies of target transformation_example
[ 33%] Linking CXX executable transformation_example
/usr/bin/ld: cannot find -llibjpeg-turbo::libjpeg-turbo
collect2: error: ld returned 1 exit status
CMakeFiles/transformation_example.dir/build.make:114: recipe for target 'transformation_example' failed
make[2]: *** [transformation_example] Error 1
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/transformation_example.dir/all' failed
make[1]: *** [CMakeFiles/transformation_example.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

qm13, 2021-06-28:
Can you please provide more information e.g SDK version, compiler etc...

xXCXMXx, 2021-07-01:
Thanks for the reply.
Runing on Jetson Xavier AGX 
SDK is 1.4.1. 
cmake version 3.20.2

I think the issue is the failure of locating or installing libjpeg-turbo package. May I know the alternative solution to it?

closed 2021-10-06 by qm13