microsoft/Azure-Kinect-Sensor-SDK #1632: compatible usb type A cables

PierrePlantard, 2021-06-29:
Hello,
I lost my official USB type A to power and USB type A to type C cables.
Does anyone have a reference for two compatible USB type A cables?

Thank you in advance,

Pierre

diablodale, 2021-07-04:
The documentation discusses this https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification#power-device
What questions more do you have?

PierrePlantard, 2021-07-05:
@diablodale Yes indeed I have read the information, such as "The power connector is a 4.5mm OD with 3.0mm ID and a pin diameter of 0.6mm". However I can't find the desired cables on sites like Amazon.
So I wanted to know if the community has any weblinks to cables that work with the kinect (Power to Usb type A and Usb type C to Usb type A).

diablodale, 2021-07-05:
Amazon has many https://www.amazon.com/s?k=usb-a+to+usb-c&ref=nb_sb_noss_2
https://www.amazon.com/s?k=usb-a+to+power+barrel&ref=nb_sb_noss_2

I recommend you ensure the size and polarity +/- is correct on the barrel cable.

closed 2021-08-23 by qm13