microsoft/Azure-Kinect-Sensor-SDK #1634: Unstable Body Tracking with some people

Rudizzzz, 2021-07-01:
**Describe the bug**

With some people we noticed that body tracking is highly unstable.
Unfortunately it doesn't depend on the environment or what the person is wearing.
So far we've tested about 30 people and 2 of them have this instability problem.
We are really worried that when we launch the device on the market many users will face this unstable body tracking.
Could you please explain better what the problem is and how to overcome it?
It would be very appreciate it.

**To Reproduce**

1. Launch 'k4abt_simple_3d_viewer.exe'
2. Test different people until you will find a person that has an unstable body tracking, even if he's standing still

**Expected behavior**

I expect that the body tracking algorithm behave the same no matter what kind of adult person is in front of the camera.

**Logs**

**Screenshots**

[image]
The person marked orange has very unstable body tracking even if he's standing still.
The person marked blue has very stable body tracking.
The problem also exists when the orange person is the only person in the FOV.
The two guys are about 30 years old.
[image]
[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 Enterprise LTSC
 - SDK Version:  Camera 1.4.1, Body Tracking 1.1.0
 - Firmware version:  RGB Camera: 1.6.110, Depth Camera: 1.6.79
 - GPU: NVidia GeForce GTX 1660 Ti

**Additional context**

I recorded a video while the 'k4abt_simple_3d_viewer.exe' was running.
The link below is about the video that shows the instability on the orange person.
https://we.tl/t-ygwBpfRDs0

rudizz, 2021-08-25:
Hello,
would it be possible to have any feedback regarding this issue?
I can provide further information about it, if it helps.
Thanks

Rudizzzz, 2022-01-26:
Hello,
is there any news about it?
Is the release of a new body tracking model planned?

Rudizzzz, 2022-07-22:
Dear support,
We are still facing the same problem with many users.
We have tried all versions, including the latest 1.1.2.
So far we have not figured out what is causing it. It seems that some kind of short trousers trigger the instability, but it is not so repeatable.
This problem does not allow us to use the Kinect4Azure camera in our commercial products, although we would like to.
Any information on this would be greatly appreciated.
Thanks

ekdrnqkdrn, 2022-07-25:
#1751

ekdrnqkdrn, 2022-07-25:
#1751 
Still We also have the same problem.

what we checked

1. When wearing a black mask
2. When to wear black
3. When your hair covers ears
4. When raising one hand or one foot after turning the body 90 degrees toward the camera
5. When lifting arms or legs vertically

We also hope that this issue will be resolved soon...

rudizz, 2022-07-25:
Thank you for your reply.
Unfortunately we have instability problems even without the conditions you mentioned.
We never use masks because we only use the camera to detect one person.
Sometimes, even standing still with the arms down, the skeleton goes crazy.
Usually the person is placed at a distance of 2.5 metres from the camera, and if he or she approaches the camera by about 0.5 metres, the instability ceases.

We found this instability in about 10 per cent of the people detected by K4A.
Have you also encountered the same percentage?
Isn't it strange that so far no one else seems to have this problem?

Rudizzzz, 2023-03-31:
Dear Support,
do you have any news or comment about this problem?
Do I have to reupload the video of the instability?
Any feedback would be greatly appreciated.