microsoft/Azure-Kinect-Sensor-SDK #1640: How can i get the information about Focus with a kinect DK?

Applied-Energetic, 2021-07-09:
I am doing some work about Multi-focus Image Fusion.And I try to get some train dataset by Kinect dk which can provide me RGB and Depth data. SO I want to know if Kinect dk can Adjust the focus of kinect dk . if not , how can I calculate the focus ?

diablodale, 2021-07-10:
It is my understanding that the Kinect Azure RGB camera is a fixed focus camera.
You can get the camera calibration matrix which contains the focal point and other intrinsic/extrinsic values using https://docs.microsoft.com/en-us/azure/kinect-dk/use-calibration-functions#retrieve-calibration-data

Applied-Energetic, 2021-07-12:
Very thank you for response，and i do get the matrix and focal point .Just like these picture.
[image]
[image]
SO I want to know is the matrix Unchangeable？If I can change it , how should I do?
It is very important for my work , if it is possible I can get the Multi-focus image as my train data.
Thanks again.

diablodale, 2021-07-12:
The calibration matrices (intrin, extrin) are unchangable for a given mode (wide pov, narrow pov, etc.). There is no variable lens focus.

Applied-Energetic, 2021-07-13:
Thank you very much.

Applied-Energetic, 2021-07-13:
problem solved

closed 2021-07-13 by Applied-Energetic