microsoft/Azure-Kinect-Sensor-SDK #1642: Running 2 instances of k4abt_tracker_t on 2 separate devices. Device 2 only reaches 15fps.

tobibaum, 2021-07-12:
Azure Body Tracking SDK: 1.1.0
Azure Kinect SDK: 1.4.0

Hey there,
We are working on multi-camera body tracking. We built a non-blocking demo on top of the simple_3d_viewer code and incorporated multi-camera synchronization. 

In pseudo-code, we do something along the lines of:
```
std::vector<k4a::device> devices(device_count);
std::vector<k4abt_tracker_t> trackers(device_count);
std::vector<int> status(device_count);
std::vector<uint64_t> last_ts(device_count);

for (i<2){
    if (status[i] == 0){
        // push a new frame to the tracker and remember capture timestamp
        k4a::capture cap;
        devices[i].get_capture(&cap, ..., 0);
        if(success){ // got a new frame from camera
            last_ts[i] = cap.get_timestamp();
            k4abt_tracker_enqueue_capture(trackers[i], cap.handle(), 0);
            status[i] = 1;
        }
    }
    if(status[i] == 1){
        k4abt_tracker_pop_result(trackers[i], &bodyFrame, 0);
        if(success){ // got a result from tracker
            // how long did the tracker take to compute this?
            time_taken_for_tracking[i].push_back(now - last_ts[i]); 
            result_queue[i].push_back(bodyFrame, last_ts[i]);
            status[i] = 0;
        }
    }
}

// synchronization and blocking!
if(....){ // all devices produce a new tracking result
    // visualize the whole thing
    // print FPS of displayed result
}
```

the first half of this code is non-blocking, as we only check whether the capture/tracker queues have new frames to process. Running this code with a single camera and instance of `k4abt_tracker_t` will work at 30fps, as expected. As soon as we add another camera and tracker, this second tracker drops to a throughput of 15fps.

Please find below the example output of our program:
```
Found 2 connected devices:
0
1
FPS: 26
FPS: 33
FPS: 32
Timestamps dev0 (102): 701 735 768 901 935 968 1001 1035 1068 1101 1135 1168 1201 1235 1268 1301 1335 1368 1401 1435 1468 1501 1535 1568 1601 1635 1668 1701 1735 1768 1801 1835 1868 1901 1935 1968 2001 2035 2068 2101 2135 2168 2201 2235 2268 2301 2335 2368 2401 2468 2501 2535 2568 2601 2635 2668 2701 2735 2768 2801 2835 2868 2901 2935 2968 3001 3035 3068 3101 3135 3168 3201 3235 3268 3301 3335 3368 3401 3435 3468 3501 3535 3568 3601 3635 3668 3701 3735 3768 3801 3835 3868 3901 3935 3968 4001 4035 4068 4101 4135 4168 4201
Timestamps dev1 (56): 468 535 668 735 801 868 935 1001 1068 1135 1201 1268 1335 1401 1468 1535 1601 1668 1735 1801 1868 1935 2001 2068 2135 2201 2268 2335 2401 2468 2535 2601 2668 2735 2801 2868 2935 3001 3068 3135 3201 3268 3335 3401 3468 3535 3601 3668 3735 3801 3868 3935 4001 4068 4135 4201
Finished body tracking processing!
0 0.237624 ms / frame || 52
1 0.128713 ms / frame || 50
10 0.049505 ms / frame || 2025
11 0.366337 ms / frame || 27982
0: time spent in status 1 32.9706
1: time spent in status 1 50.5893
```

we plot at 30 FPS, because camera0 spits out new tracking results at that rate. nice! Unfortunately, camera1 does not keep up with this rate. as you can see in the above log, there are 102 timestamps for tracking results of cam0, whereas there are only 56 for cam1.
Indeed, when we look at the time each of the cameras spent in status 1 (wait for tracking results), we see that camera0 takes 32.97ms (perfect!), but camera1 takes 50.59ms.

is there a different way of running two trackers at the same time? As it is right now, we will not be able to run an application at the desired 30fps. let alone the planned 60fps we plan for incorporating additional 2 cameras.

The GPU usage is at only 40%. I am pretty sure that the deep learning model in onnx should be able to handle a larger batchsize. Is there anything in the pipeline to have combined trackers that share the call-out to the GPU to speed things up in this domain?

My code is based off of the simple_3d_viewer example, but ported to cmake in order to use the body tracking SDK 1.1.0 (which lets me specify the model path and use my RTX3090):
https://github.com/tobibaum/AzureKinectSample/blob/tobi/sample/cpp/skeleton/main.cpp#L310-L437

Thanks a lot!!
Tobi

diablodale, 2021-07-12:
Some ideas

- have you correctly setup the strobing time offsets so that the sensors do not conflict with each other? Are you sure?
- it is unclear the scope of `cap` and `bodyframe`. Are these local and thread-safe variables?
- Are you checking return results from calls like `k4abt_tracker_enqueue_capture()`?
- While its ok to mix c/c++, you might save later debugging to keep with one or the other. `k4abt_tracker_enqueue_capture` -> `k4abt::tracker::enqueue_capture`
- I recommend you isolate source of the lower fps. What call is resulting in less than 30fps? Are you not getting 30fps device captures? Not getting 30fps tracker.enqueues? Not getting 30fps tracker.pops? etc.
- Do you have different results using different tracker compute devices (cpu, cuda, directml, tensorrt)?

tobibaum, 2021-07-14:
Thank you so much for taking this on and your very helpful thoughts :)

just to briefly address all of your bullets:
- I would rule out effects of the active IR depth sensing. I get similar results no matter where cameras point
- I wasn't quite clear on the effects of cap and bodyframe, so I made everything vectors, with `device_count` entries.
- I am now checking the output of the enqueue op (see below)
- rewrote the whole thing in C.
- great point!! I did a new analysis where I count the occurrence of each operation per second. this give a very clear indication, where we lose speed.
- I am not quite sure how to switch out the default for a different compute device :/ the only thing I could think of was the `tracker_config.processing_mode = CPU`, which of course resulted in way lower frame rates. I tried the same under windows and ubuntu. same problems.

Now to get into more detail about my latest investigations. I rewrote the code in an as minimal as possible manner:

Initialization code:
```
uint32_t device_count = k4a_device_get_installed_count();                    
printf("Found %d connected devices:\n", device_count);                       
                                                                             
std::vector<k4a_device_t> devices(device_count);                             
std::vector<k4abt_tracker_t> trackers(device_count);                         
int depthWidth, depthHeight;                                                 
std::vector<k4a_calibration_t> sensorCalibrations(device_count);             
                                                                             
for(uint8_t dev_ind = 0; dev_ind < device_count; dev_ind++){                 
    k4a_device_t dev = nullptr;                                              
    VERIFY(k4a_device_open(dev_ind, &dev), "Open K4A Device failed!");       
                                                                             
    // Start camera. Make sure depth camera is enabled.                      
    k4a_device_configuration_t deviceConfig = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
    deviceConfig.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;                  
    deviceConfig.color_resolution = K4A_COLOR_RESOLUTION_OFF;                
                                                                             
    k4a_calibration_t sensorCalib;                                           
    VERIFY(k4a_device_start_cameras(dev, &deviceConfig), "Start K4A cameras failed!");
    VERIFY(k4a_device_get_calibration(dev, deviceConfig.depth_mode, deviceConfig.color_resolution, &sensorCalib),
        "Get depth camera calibration failed!");                             
                                                                             
    depthWidth = sensorCalib.depth_camera_calibration.resolution_width;      
    depthHeight = sensorCalib.depth_camera_calibration.resolution_height;    
                                                                             
    // Create Body Tracker                                                   
    k4abt_tracker_t track = nullptr;                                         
    k4abt_tracker_configuration_t tracker_config = K4ABT_TRACKER_CONFIG_DEFAULT;
    tracker_config.processing_mode = K4ABT_TRACKER_PROCESSING_MODE_GPU;      
    VERIFY(k4abt_tracker_create(&sensorCalib, tracker_config, &track), "Body tracker initialization failed!");
                                                                             
    devices[dev_ind] = std::move(dev);                                       
    trackers[dev_ind] = std::move(track);                                    
    sensorCalibrations[dev_ind] = std::move(sensorCalib);                    
}                                                                            
                                                                                                                                                                                        
uint32_t fps = 0, fps_counter = 0;                                           
auto start_time = std::chrono::high_resolution_clock::now();                 
auto absolute_start_time = std::chrono::high_resolution_clock::now();        
auto now = std::chrono::high_resolution_clock::now();                        

std::vector<std::map<std::string, int>> counters(device_count);              
                                                                                
std::vector<k4abt_frame_t> bodyFrames(device_count);
std::vector<k4a_capture_t> sensorCaptures(device_count);        
```

The main loop and interesting part!
```
while (s_isRunning)                                                          
{
    for(int dev_ind = 0; dev_ind < device_count; dev_ind++){
        // ==== 1. get a new capture =====
        k4a_wait_result_t getCaptureResult = k4a_device_get_capture(devices[dev_ind], &sensorCaptures[dev_ind], 0); 
        if (getCaptureResult == K4A_WAIT_RESULT_SUCCEEDED)                   
        {                                                                    
[... 131 lines cut ...]
enq 29
pop 29
==1==
cap 30
enq *15*
pop *15*
--FPS: 29--
==0==
cap 31
enq 29
pop 29
==1==
cap 31
enq *16*
pop *17*
```

camera1 enqueues and pops 15~18 times per second, which would lead to a lower frame when drawing combined results.

[... cut ...] h competition of GPU resources? Unfortunately I cannot test this with a 2GPU setup.

How does the bodytracker work internally? Does it call out to the deep learning model as a batch? Do you have any other ideas on what to change here and how to further isolate the problem? I feel like I am at a stage with this bug, where I would have to peak into what is happening between `k4abt_tracker_enqueue_capture` and `k4abt_tracker_pop_result`, which is of course impossible :/

Thanks for your help!!
tobi

tobibaum, 2021-07-14:
please find attached the code that reproduces this behavior.
It is based on `simple_3d_viewer`

https://github.com/tobibaum/Azure-Kinect-Samples/tree/3f26d5fda5615181a79a91f1b86d9ea1619a624f

compiled on Ubuntu18 with:
```
mkdir build
cd build
cmake .. -GNinja
ninja
```
run:
```
cd bin
./simple_3d_viewer
```

tobibaum, 2021-07-15:
I additionally performed the following tests:

I rewrote `simple_3d_viewer` to accept a command line argument on which of the 2 attached cameras to use (0 or 1).

I can now run
```
./simple_3d_viewer 0
```
or
```
./simple_3d_viewer 1
```
and each program will run the code from above (simple capture, enqueue to tracker, pop from tracker, throw away the results) on one of my two cameras.

Separately, each of these calls results in `cap`, `enq` and `pop` of 30 times per second (as expected). As soon as I turn on the second program, the `enq` of the running program drops, as well as the count for the secondary program stays low.

The two independent programs each show up in `nvidia-smi` with ~1GB GPU memory usage. For just a single program, GPU-utility is at ~50%. for two programs this only rises to ~86%. I am currently testing on an NVIDIA RTX2070s, CUDA 11.0 with 8GB of GPU memory.

diablodale, 2021-07-17:
Hi. Some thoughts. Please take them as opinions and intended to be constructuve.

* I don't have currently have 2 Azure kinect's. Nor Ubuntu. So I can't directly test your code.
* I see several coding issues related to your mixing of C and C++. For example, you often use std::move() in places where it will not function as you likely intend. https://en.cppreference.com/w/cpp/language/move_assignment#Trivial_move_assignment_operator
* Other times I'm concerned there is perhaps a handle leak/lost in some situations due similar mixing/misuse. Given what I see, I recommend you use C++ only so that construct/destruct/releases are handled automatically for you. A quick check is that if any Kinect variable type or function call has an underscore in it...you are probably using C instead of C++.
* Also appears to be direct coding errors like the block starting with `if(device_count>1 && bodyFrames[1] != nullptr){`. I think that should be instead `if(device_count>1 && bodyFrames[dev_ind] != nullptr){` and same in lines below.
* Have you read every word of this page... including this section for Ubuntu https://docs.microsoft.com/en-US/azure/Kinect-dk/multi-camera-sync#linux-computers-usb-memory-on-ubuntu
* Do you have a dedicated USB host controller for every Kinect?
* Can you reproduce your issue by running the demo apps that Microsoft provides pre-built? For example, if you do all the above config things, point your Kinects in a way it is impossible for them to conflict, then run the `k4aviewer` binary as provided and precompiled by Microsoft. Run them, choose the separate Kinect, and run unsynced -- no master/slave. Do you get 30fps? Then run them configured with master/slave with needed timing config. Do you get 30fps?

I have a RTX2070Super (though it is on my laptop). If I use CUDA, I see no way I could run two Kinect body trackers. If I use Direct ML, my GPU has bandwith but Microsoft's code is single-threaded somewhere and one of my cores is pegged >95%, so again I couldn't run two Kinect. This is on Windows, so Ubuntu might have other behavior.

diablodale, 2021-07-18:
Just remembered that when you have multiple Kinect in master/slave, you *must* enable the color camera on the master camera. You don't have to get the color data from a capture...you only have to enable the color camera as your start it. This is because the entire sync process is directly related to the VSYNC of the color camera on the master. And you also want to have manual exposure on that Kinect so that the timing of this sync signal isn't affected by the master camera auto adjusting exposure therefore changing the sync.

This is the current design of the SDK and there is no alternative at this time. You can enable it with low processing cost with `K4A_COLOR_RESOLUTION_720P` and `K4A_IMAGE_FORMAT_COLOR_NV12`

This is something to keep in mind as you further develop your solution.

tobibaum, 2021-07-19:
Hey Dale,

Thank you very much more for your comments and ideas to fix my problem :)

I agree, that my style is not necessarily the cleanest :D I rewrote everything with only the C++ versions of the library calls. I slimmed down the code even further and am sure, that none of the following are the reason for the delays:
- USB connections
- USB grub driver settings
- master/subordinate settings
- windows/linux

As I mentioned, I can always capture frames from the devices at 30fps. I measure this by counting the "cap", "enq", "pop" events every second, as described in my previous post.
My code starts dropping frames in the `k4abt::tracker` part of things. Enqueuing and popping captures isn't accomplished 30 times per second.

Your recent comment:
> I have a RTX2070Super (though it is on my laptop). If I use CUDA, I see no way I could run two Kinect body trackers. If I use Direct ML, my GPU has bandwith but Microsoft's code is single-threaded [...]

helped me get on the right track. I suspected all along, that the library call to k4abt is doing something less than optimal. I know that the DL-model itself should be able to handle large batches of captures, but the inference code itself only seems to run on a single frame in a single-threaded fashion, using up a large part of the GPU.

**Solution:**
Since you had mentioned **tensorrt** before, I read up on it a little bit more and installed it on my machine. I am now able to run 30 enqueues and pops to the `k4abt::tracker` on both cameras. 

@qm13 , does your internal team use tensorrt to runs 3 trackers at the same time, as you mention in this post? https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1358#issuecomment-696253380
If not, how did you manage to do this using CUDA or DirectML?
Thanks for briefly confirming this :)

I still believe that the tracker functionality could be drastically improved by having some way of batching up GPU calls. I understand that this might not be a priority, since it might be an unusual edge-case. Eventually, we want to run 4 Azure Kinects at 30fps and display results in real-time.
Would be a great feature down the line.

Thanks!
Tobi

diablodale, 2021-07-19:
Great progress :-).  FYI, current k4abt sdk on Windows has bug in the interaction between ONNX and TensorRT which leads to a big memory leak. Microsoft isolated it and wrote a fix is planned for the next major sdk update. Ubuntu might be different https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1576

Curious...immediately above you are using RTX2070 while in OP were using RTX3090. Do you see any clear patterns or changes in capabilities between the two in your test case?

tobibaum, 2021-07-22:
thanks for noticing and asking :)

I have indeed access to both cards. The effect is similar but less pronounced for the rtx3090.
Whereas the 2070s drops the processing rate to 15 fps for the second camera, on the 3090 it only drops to 25 fps.

I assume the underlying problem is the same and we were able to respectively circumvent it using TensorRT.

Thanks!
Tobi

closed 2021-07-22 by tobibaum