microsoft/Azure-Kinect-Sensor-SDK #1644: Dependencies table for ONNX Runtime is wrong

UnaNancyOwen, 2021-07-19:
**Describe the bug**

Is the dependencies table for ONNX Runtime in [this document](https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#required-dlls-for-onnx-runtime-execution-environments) correct?
I think that's not correct, at least not when it comes to DirectML. (directml.dll is missing.)
It is also strange that CUDA DLLs are included in the dependencies of CPU and TensorRT.
https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#required-dlls-for-onnx-runtime-execution-environments

**Screenshots**

[image]

**Additional context**

diablodale, 2021-07-20:
related to https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1541

qm13, 2022-03-21:
Fixed

closed 2022-03-21 by qm13