microsoft/Azure-Kinect-Sensor-SDK #1645: I am looking for longer cables for Azure Kinect cameras, preferably 3 - 5 m. I am using two kinects to reconstruct the scene and I want to place the cameras at up to 6 - 8 m distance.  I have tried using this Oculus Quest Link Cables https://www.amazon.de/KIWI-Design-Oculus-Daten%C3%BCbertragung-Hellgrau-Hellgrau-3M/dp/B08BPDVBHF/ref=sr_1_22?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=Oculus%2Bconnection%2Bcable%2C%2B3%2Bm%2C%2Bhigh- , but I am not able to start both cameras using these cables.   Is there any specific type of cable (preferably available in the EU) that works without any issues?

crypto-g245, 2021-07-19:
#### Application:

- Using two kinects to reconstruct the scene 

#### Configuration:

- Extend the data cable up to 3 meters using an Active USB cable (no extenders). Used cable to connect the cameras - https://www.amazon.de/KIWI-Design-Oculus-Daten%C3%BCbertragung-Hellgrau-Hellgrau-3M/dp/B08BPDVBHF/ref=sr_1_22?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=Oculus%2Bconnection%2Bcable%2C%2B3%2Bm%2C%2Bhigh-

- Using original power cables supplied with the device
- Azure kinect SDK V 1.4.1 (https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md)

#### Issues:
-  One of the cameras is streaming the data correctly. but the second camera would display the errors in the log as displayed in the below snapshot from Kinect Viewer and the camera stops working with the following errors in series. The device is recognized when connected. Once the camera is started in Kinect viewer, the device stops working after few seconds. 

-  Both the device and cables are working and there are no issues with them. 
    
[image]

- The camera stops working after this error and Azure Kinect viewer displays following errors. 

[image]
[image]
[image]
[image]
[image]

  
- Both the cameras are working fine with the original type-c to type-A cables. This issue doesn't occur with original cables.

_Originally posted by @crypto-g245 in https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1473#issuecomment-876426004_

UnaNancyOwen, 2021-07-19:
You may want to consider using this cable. (Unfortunately, the cables that are longer than standard are special and very expensive.)
https://www.newnex.com/technology-azure-kinect-extension.php

crypto-g245, 2021-07-20:
Hi @UnaNancyOwen ,
Have you already tested this cables with Azure Kinect Sensors?

UnaNancyOwen, 2021-07-20:
Yes, it works fine.

crypto-g245, 2021-07-20:
Hi @UnaNancyOwen ,
Thanks for quick response. 
Could you please elaborate on length of the cables you used?

I'm little skeptical,  Because Microsoft mentioned one should use active USB if it's longer than 1.5 m where as these cables (only upto 5m) are passive.

Will these cables less than 5m also work fine with these Sensors?

UnaNancyOwen, 2021-07-20:
I have used these mid-range cables (8m, 16m). these cables are active cable.
This manufacturer has confirmed that it works with Azure Kinect. It is the only manufacturer that I know of.
https://www.newnex.com/usb-3-1-type-c-active-cable-legacy.php
https://www.newnex.com/usb-3-1-type-c-active-cable.php

crypto-g245, 2021-07-20:
Thank you @UnaNancyOwen !

guss2149, 2021-08-19:
Hello @UnaNancyOwen.  I am wondering what the USB-A cable you linked was hooked up to is? And did you use the separate power supply to power the camera? The camera doesn't get enough power through a USB-A cable so a separate power supply is needed. We tried the USB-C to USB-C cable you linked plugged into the Desktop that does not have USB-C on the motherboard but I tried a PCI-E USB-C expansion card with no success. The USB chipset is apparently not compatible with the camera and I can't find one with a chipset and is PCI-E and is sold in the US that is compatible with the camera. We are really struggling with finding an alternate power source other than the 110V power supply. We need an alternate solution because the cable lengths are so long. Also using an active cable doesn't work for power either.

I am very curious what you did to power the camera.

UnaNancyOwen, 2021-08-20:
> I am wondering what the USB-A cable you linked was hooked up to is? 

I needed to connect multiple Azure Kinects. However, one PC had a limited number of USB C ports, so I used USB A ports.

> And did you use the separate power supply to power the camera?

Yes, I used additional power supply. That is, we used two cables.
I extended it with [extension cord](https://en.wikipedia.org/wiki/Extension_cord), with USB power charger that included in Azure Kinect DK box.

> We tried the USB-C to USB-C cable you linked plugged into the Desktop that does not have USB-C on the motherboard but I tried a PCI-E USB-C expansion card with no success. 

[UL-C21C11](https://www.newnex.com/usb-3-1-type-c-active-cable.php) does not support power supply only one cable. An additional power supply is required.
However, [UL-C21C31](https://www.newnex.com/azure-kinect-extension-cable.php) supports power supply only one cable. Please contact manufacturer for details.

These are special cables. Please note that these cables are directional.
If these cables are plugged into the device and host in opposite directions, they cannot be used.
In the first time, I got it wrong. Please check "To Host" (blue plug) into Azure Kinect port, and "To Device" (black or white plug) into PC port.

> The USB chipset is apparently not compatible with the camera and I can't find one with a chipset and is PCI-E and is sold in the US that is compatible with the camera. 

Unfortunately, I have not used the USB C expansion card, so I can't introduce it.
For USB A, I used this [one](https://www.startech.com/en-us/cards-adapters/pexusb3s44v).

guss2149, 2021-08-24:
Hello. Thank you for your response. I appreciate the info. We purchased the UL-C21C31-10M but the PC we are using doesn't have USB-C and the PCI-E card's USB Host Chipset (ASMedia) isn't compatible with the Kinect Azure Cameras. We keep getting errors and cameras disconnected, needing to be unplugged and plugged back in, on one of our PCs (it has two cameras plugged into it) and were not sure if it's the power or the comms. 

I'm going to try the PCI-E card you linked and the FIRENEX-ULC-8 cables next. Thanks again for your help.

closed 2021-10-06 by qm13