microsoft/Azure-Kinect-Sensor-SDK #1648: Azure depth platform consumer SDK?

dimitrijevic, 2021-07-21:
Could someone please update on the state of the Azure depth platform "consumer" Sensor SDK...would like to go into a development of a product using Azure Kinect, but prior this year there was talk there would be other devices soon and Azure Kinect would be getting another "consumer" SDK...is that still coming, and in what form, given Azure Sendor Kinect SDK is multi-platform will Microsoft be able to support multiple platforms with such a "consumer" SDK or will that be Win(11)-only and possibly backcompatible with the previous Windows.Media.Capture.Frames usecases?

Anything more that could be given would be appreciated...even if it's cancelled.

If you can't provide any more Kinects (given the store call to order is grayed out and the call center has no idea when it will be made available again only hoping it's end of this month maybe) which other device should we be looking at please?

It would be great if it were available in the EU/Germany besides US but I know I'm hoping too much.

qm13, 2021-10-06:
Yes, it is still coming. The SDK is an evolution of the current SDK with minor breaking changes. You should not have any problems migrating from the current SDK to v2.0 (when it comes).

closed 2021-10-06 by qm13

dimitrijevic, 2022-03-02:
Please any news on this?