microsoft/Azure-Kinect-Sensor-SDK #1649: Kinect dependencies not available for download - long-term support again failing

diablodale, 2021-07-21:
Dependencies for the Kinect v1 and v2 SDKs have disappeared from the official download locations at Microsoft Downloads. The links throughout various Microsoft websites and Microsoft download server lists don't work.

https://developer.microsoft.com/en-us/windows/kinect/
scroll down and click on `Microsoft Speech Platform SDK 11`
It will report `We're sorry, this download is no longer available.`

https://www.microsoft.com/en-us/download/search.aspx?q=speech+recognition&first=1
click on any of the Speech Platform v11 downloads. All will report `We're sorry, this download is no longer available.`

Both the Microsoft Speech Platform SDK v11 and Microsoft Speech Platform Runtime v11 are needed for full development and usage of Kinect v1 and v2 SDKs.

This is another example of https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1083

qm13, 2021-08-16:
The Azure Kinect team does not own the previous Kinect products. I have passed the issue to the team that owns these products which are now end of life products.

closed 2021-08-16 by qm13

diablodale, 2021-08-16:
With what process or person "to the team" can I associate? I ask so to track progress and close loops on success or failure.

qm13, 2021-08-16:
@diablodale if you have not already done so you should file a bug via Windows Feedback Hub at
https://microsoft.sharepoint.com/sites/mywork/SitePages/Windows10/Feedback-Support.aspx

diablodale, 2021-08-18:
Hi, is that a site generally available to customers? For 2 days I've tried to get to it and it always reports

```
hat didn't work
We're sorry, but dale@hidale.com can't be found in the microsoft.sharepoint.com directory. Please try again later, while we try to automatically fix this for you.
```