microsoft/Azure-Kinect-Sensor-SDK #1651: sample freezes on k4abt_tracker_create... any ideas why?

skywalkerwin, 2021-07-31:
Hello I am trying to run a body tracking sample and it seems like somethings getting stuck here:

[image]

Kinect Sample can be found here: https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download

Any ideas why the sdk would freeze trying to create a tracker?

Thanks you!!

qm13, 2021-08-16:
Can you run the pre-compiled k4abt_simple_3D_viewer on your system?

skywalkerwin, 2021-08-16:
> Can you run the pre-compiled k4abt_simple_3D_viewer on your system?

I can

closed 2021-10-06 by qm13

peterkennard, 2021-12-03:
same issue here windows 10 Pro installed and the latest stable kinect SDK and body tracker SDKs
I built the simple_3d_viewer from master with VisualStudio 1019 , I updated nuget packages, built both debug and release versions,
Tried to run it ang hangs in k4abt_tracker_create no error messages, just hangs.
I am running it in the CPU mode, and do not have a kinect (HW) installed on the machine.

The binary supplied with the SDK runs fine in GPU mode.

The program I built will run with the test.mkv file just very slowly.  It will not run in GPU mode.

The machines has a RTX 3060 Laptop GPU on it and is set ro use that for the desktop.  Other GPU programs work fine.

submorphic, 2021-12-23:
Having similar problem.  simple viewer shipped with SDK runs fine on three different laptops.  When I build a simple viewer kit and run it is fine on two laptops, but hangs without error on k4abt_tracker_create() on the third laptop. I'm capturing trace info and will update post.

peterkennard, 2021-12-24:
Thanks would be great to solve this 😊

harryspiercy, 2022-02-10:
Hi all, I encountered the same issue while running the latest SDK from my laptop with a RTX 3060.

I haven't found a proper solution but have found changing the trackers config processing mode to CPU stops the hanging when hitting the create tracker method.

`trackerConfig.processing_mode = K4ABT_TRACKER_PROCESSING_MODE_CPU;`

Would be great to get a solution for running with the GPU of course but thought this may help someone get going!

thisisfangsheng, 2022-02-22:
Hello, I run into the same problem. Looking forward to the solution. Thanks.