microsoft/Azure-Kinect-Sensor-SDK #1653: Unable to get color images using custom code

PucklaMotzer09, 2021-08-02:
**Describe the bug**
I want to write some code which uses the SDK, but for some reason I am unable to get a color image from the camera, but the Kinect Azure Viewer works completely fine.

**To Reproduce**
Here is a code sample, which uses SDL to display the image:
```cpp
#include <SDL2/SDL.h>
#include <SDL2/SDL2_framerate.h>
#include <cassert>
#include <iostream>
#include <k4a/k4a.hpp>
#include <k4abt.hpp>

int main(int args, char *argv[]) {
  assert(SDL_Init(SDL_INIT_VIDEO) == 0);

  auto *window =
      SDL_CreateWindow("NoColorFrame", SDL_WINDOWPOS_CENTERED,
                       SDL_WINDOWPOS_CENTERED, 1280, 720, SDL_WINDOW_SHOWN);
  assert(window);
  auto *renderer = SDL_CreateRenderer(window, -1, SDL_RENDERER_ACCELERATED);
  assert(renderer);
  auto *texture = SDL_CreateTexture(renderer, SDL_PIXELFORMAT_YUY2,
                                    SDL_TEXTUREACCESS_STREAMING, 1280, 720);
  assert(texture);
  FPSmanager fps_manager;
  SDL_initFramerate(&fps_manager);
  SDL_setFramerate(&fps_manager, 30);

  auto device(k4a::device::open(K4A_DEVICE_DEFAULT));
  k4a_device_configuration_t device_config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
  device_config.camera_fps = K4A_FRAMES_PER_SECOND_30;
  device_config.color_format = K4A_IMAGE_FORMAT_COLOR_YUY2;
  device_config.color_resolution = K4A_COLOR_RESOLUTION_720P;
  device_config.synchronized_images_only = false;
  device_config.depth_mode = K4A_DEPTH_MODE_OFF;
  device.start_cameras(&device_config);

  bool running{true};
  SDL_Event event;

  while (running) {
    while (SDL_PollEvent(&event)) {
      switch (event.type) {
      case SDL_QUIT:
        running = false;
        break;
      }
    }

    {
      k4a::capture capture;
      if (device.get_capture(&capture, std::chrono::milliseconds(1))) {
        if (const auto color(capture.get_color_image()); color) {
          assert(SDL_UpdateTexture(texture, nullptr, color.get_buffer(),
                                   color.get_stride_bytes()) == 0);
        }
      }
    }

    SDL_RenderCopy(renderer, texture, nullptr, nullptr);
    SDL_RenderPresent(renderer);

    SDL_framerateDelay(&fps_manager);
  }

  SDL_DestroyTexture(texture);
  SDL_DestroyRenderer(renderer);
  SDL_DestroyWindow(window);
  SDL_Quit();
[... 76 lines cut ...]
[2021-08-02 11:42:15.805] [trace] [t=9428] d:\a\1\s\extern\azure-kinect-sensor-sdk\src\usbcommand\usb_cmd_priv.h (101): usbcmd_t_destroy(). Destroyed usbcmd_t 0000017685449D70
[2021-08-02 11:42:15.805] [trace] [t=9428] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color_mcu\color_mcu.c (41): colormcu_t_destroy(). Destroyed colormcu_t 00000176853B60F0
[2021-08-02 11:42:15.805] [trace] [t=9428] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (53): k4a_device_t_destroy(). Destroyed k4a_device_t 00000176854486C0
[2021-08-02 11:42:15.807] [trace] [t=9428] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\dynlib\dynlib_windows.c (22): dynlib_t_create(). Created   dynlib_t 000001768549AF50
[2021-08-02 11:42:15.810] [info] [t=9428] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\dynlib\dynlib_windows.c (131): dynlib_create(). Depth Engine loaded C:\Projekte\k4abt_no_color_image\build\windows\x64\debug\depthengine_2_0.DLL
[2021-08-02 11:42:15.810] [info] [t=9428] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\deloader\deloader.cpp (51): verify_plugin(). Loaded Depth Engine version: 2.1.3
[2021-08-02 11:42:15.811] [trace] [t=9428] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\dynlib\dynlib_windows.c (22): dynlib_t_destroy(). Destroyed dynlib_t 000001768549AF50
```

**Desktop (please complete the following information):**
 - OS with Version: Windows 10
 - SDK Version:  1.4.1
 - Firmware version:  *see logs*

**Additional context**
If I enable depth I get a lot of these lines:
```
[2021-08-02 11:44:47.598] [error] [t=16316] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1527733 type:Depth
[2021-08-02 11:44:47.632] [error] [t=16316] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1561055 type:Depth
[2021-08-02 11:44:47.665] [error] [t=16316] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1594400 type:Depth
[2021-08-02 11:44:47.699] [error] [t=16316] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1627722 type:Depth
[2021-08-02 11:44:47.732] [error] [t=16316] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1661066 type:Depth
[2021-08-02 11:44:47.765] [error] [t=16316] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1694400 type:Depth
[2021-08-02 11:44:47.799] [error] [t=16316] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1727733 type:Depth
```

PucklaMotzer09, 2021-08-02:
Every Kinect Azure application I compile is unable to get a color image. But every precompiled application I got when downloading the SDK works as expected

qm13, 2021-08-02:
Can you confirm that the k4aviewer and k4abt_simple_3d_viewer included /tools in the installer work?

PucklaMotzer09, 2021-08-03:
Yes they both work completely fine. Only the programs which I compile myself do have this problem