microsoft/Azure-Kinect-Sensor-SDK #1654: Multiple devices / Synchronization

jdeeee, 2021-08-03:
Hi there!
I have a couple of questions regarding the synchronization of 2 or 3 kinect for azures. 

1. Up to now I have used two cameras with overlapping field of view with a connecting sync-cable, one sensor configured as master the other as subordinate and a "Delay off master" of 0. In some scenarios I have seen interference then and in some this setup seemed to have removed interference in comparison to not using a sync-cable at all. After having read the documentation multiple times now, I would actually assume that using this setup with "Delay off master"=0 should not have any effect and my observations are mainly by chance, is that correct?
2. [Here](https://docs.microsoft.com/en-us/azure/Kinect-dk/record-external-synchronized-units) it is stated that _All subordinate devices must be started before the master device._ Why is that? Does that hold in general or is that a special precaution for the recorder? 
3. Am I right that when using 3 sensors "Delay off master" for the first subordinate should be 160 and for the second 320 (or is 160 and 160 correct?)?

Thanks for the clarifications!

saulthu, 2021-08-06:
Rather than using the config option subordinate_delay_off_master_usec which sets a delay for the color camera capture, you may want to be using the option depth_delay_off_color_usec which is designed to prevent depth camera interference. The values to be used with this are separated by 160 usec, and can be positive or negative to distribute the devices around the master camera capture time. I use 8 cameras synced with overlapping FOV and using this setting prevents all inference. For 8 cameras I use values [-560, -400, -240, -80, 80, 240, 400, 560].

saulthu, 2021-08-06:
I read somewhere that all subordinate devices must be started after the master to allow sync to start up correctly. I don't think it has anything to do with recording as such, just the sync between devices.

qm13, 2021-08-06:
@saulthu thanks for your excellent answer. I can confirm that the subordinates should be started before the master.

jdeeee, 2021-08-09:
Thanks for your answers! I have two follow up questions though:

1. Is there any insight on why the subordinates should be started before the master? I thought a subordinate only starts capturing images once the master has send a trigger signal, so I would assume it does not matter, which camera is started when.
2. Am I right, when I say that `subordinate_delay_off_master_usec `  is the delay between the color camera capture of the subordinate to the master and `depth_delay_off_color_usec ` is the delay between color and depth image within one sensor? From my understanding and [here](https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync) both parameters can be used to prevent depth camera interference. However, `subordinate_delay_off_master_usec ` will cause a delay between the color cameras of both cameras (and depth and color are simultaneous within one sensor) and  `depth_delay_off_color_usec ` will cause both color images to be captured at the same time and the depth image of the subordinate to be a little bit delayed.

saulthu, 2021-08-09:
1. I don't have a precise answer for this. The same document you linked says, "Important. To get precise image capture alignment between all devices, you have to start the master device last." Since the subordinates rely on the sync signal coming from the master before they'll produce frames, I don't know why the start order matters. I make sure the subordinates are started by checking the sync jack status and start all the devices that have a sync input before starting the one device that does not have a sync input.
2. I think you're right, your understanding sounds correct. I guess the choice of sync offsets depends on whether you desire more the accurate sync of color or depth images. The `depth_delay_off_color_usec` offsets the depth camera by max +-0.8 ms, and it can be a signed offset centered on the master color capture. The `subordinate_delay_off_master_usec` must be a positive offset and can be up to a full frame period, which e.g. could be used to increase the frame rate of observations on an overlapping FOV. I think for small offsets they're much the same effect though.

qm13, 2021-08-16:
Regarding 1) The subordinates sync to the output pulse from master. With that pulse they start with a timestamp of zero. So the timestamp from 2 camera’s will be nearly identical. If the master starts first, its timestamp counts from zero almost immediately, the subordinates would reset to zero later and therefore wouldn’t be synchronized.

closed 2021-10-06 by qm13