microsoft/Azure-Kinect-Sensor-SDK #1655: Enqueue capture in playback prevent to close application

nicolas0carre, 2021-08-03:
**Describe the bug**
During the playback of a .mkv file, at the moment where we enqueue a capture (k4abt_tracker_enqueue_capture), the program can't stop after that.
No problem with processing after enqueue the capture, and of course, we release/close everything when done.
The only way to close the app is by using the task manager.
Also, this bug has appeared since we integrated the lite model and all associated files, before this update, all worked fine.

**To Reproduce**
- Open playback
- Set playback color conversion
- Get playback calibration
- Create config body tracker
- Create body tracker
- Get playback next capture
- Enqueue capture tracker
After this step, only task manager can close the app

**Expected behavior**
The app should be closed

**Desktop (please complete the following information):**
 - OS Windows 10, Version 21H1
 - SDK Version:  1.4.1
 - Firmware version:  AzureKinectDK_Fw_1.6.110079014

scotthsu98052, 2021-08-09:
_Re: ... , and of course, we release/close everything when done. ..._

Did the code call k4abt_tracker_shutdown before calling release/close?

nicolas0carre, 2021-08-10:
Yes we did

scotthsu98052, 2021-08-12:
I used [offline_processor sample code](https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/offline_processor) to repro the problem you reported. 

The only change I added to the sample code is to use the lite model as you described.
`tracker_config.model_path = "dnn_model_2_0_lite_op11.onnx";`

The program can exit successfully.
`e:\Azure-Kinect-Samples\body-tracking-samples\offline_processor\build\bin\Release> offline_processor.exe  e:\src\quentin.mkv 1.log`
`Tracking e:\src\quentin.mkv`
`frame 451`
`DONE`
`Total read 452 frames`
`Results saved in 1.log`

Or I can hit ctrl-c to terminate it.
`e:\Azure-Kinect-Samples\body-tracking-samples\offline_processor\build\bin\Release> offline_processor.exe  e:\src\quentin.mkv 1.log`
`Tracking e:\src\quentin.mkv`
`^Came 22`

Could you please also try offline_processor sample code to see if you still repro the problem?

Here are the loaded BT dependent files (I excluded system files):
[image]