microsoft/Azure-Kinect-Sensor-SDK #1660: EULA agreement cannot be asked

mgovers-philips, 2021-08-10:
**Describe the bug**

When trying to install `libk4abt1.1-dev` in Github Actions (`ubuntu-18.04` image, with the purpose of Cloud build), the installation fails with the following error message:

```txt
Preparing to unpack .../libk4abt1.1_1.1.0_amd64.deb ...

eula agree question could not be asked
libk4abt1.1 will not be installed

dpkg: error processing archive /var/cache/apt/archives/libk4abt1.1_1.1.0_amd64.deb (--unpack):
 new libk4abt1.1 package pre-installation script subprocess returned error exit status 2
Selecting previously unselected package libk4abt1.1-dev.
Preparing to unpack .../libk4abt1.1-dev_1.1.0_amd64.deb ...
Unpacking libk4abt1.1-dev (1.1.0) ...
Errors were encountered while processing:
 /var/cache/apt/archives/libk4abt1.1_1.1.0_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Installation of `k4a-tools` and `libk4a1.4-dev`, on the other hand, works just fine.

**To Reproduce**

On every push / workflow run, `libk4abt1.1-dev` is successfully installed.

**Logs**

```txt
Preparing to unpack .../libk4abt1.1_1.1.0_amd64.deb ...

eula agree question could not be asked
libk4abt1.1 will not be installed

dpkg: error processing archive /var/cache/apt/archives/libk4abt1.1_1.1.0_amd64.deb (--unpack):
 new libk4abt1.1 package pre-installation script subprocess returned error exit status 2
Selecting previously unselected package libk4abt1.1-dev.
Preparing to unpack .../libk4abt1.1-dev_1.1.0_amd64.deb ...
Unpacking libk4abt1.1-dev (1.1.0) ...
Errors were encountered while processing:
 /var/cache/apt/archives/libk4abt1.1_1.1.0_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04
 - SDK Version: 1.4.0 (also reproed on other versions) 
 - Firmware version: N/A 

**Additional context**

Github Actions Cloud Build

scotthsu98052, 2021-08-12:
It seems you install k4abt in a non-interactive environment. Please see this [thread ](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1190) to work around the EULA acceptance issue.

mgovers-philips, 2021-08-17:
My apologies for getting back to you only now. This worked, even using `apt-get` instead of `deb`, thank you!

```bash
apt-get -qq update
echo 'libk4abt1.1	libk4abt1.1/accepted-eula-hash string 03a13b63730639eeb6626d24fd45cf25131ee8e8e0df3f1b63f552269b176e38' | debconf-set-selections
apt-get -qq install -y k4a-tools libk4a1.4-dev libk4abt1.1 libk4abt1.1-dev
```

closed 2021-08-17 by mgovers-philips