microsoft/Azure-Kinect-Sensor-SDK #1661: Calibration

Kanish2002, 2021-08-10:
How to calibrate and focus particular area using azure kinect.???

qm13, 2021-08-12:
Could you please explain exactly what you want to achieve.

Kanish2002, 2021-08-13:
Sir I need to calibrate the camera for a particular area and need to give
colormap for different depths.

Like AR sandbox by UCdavis Project

Kanish2002, 2021-08-13:
https://youtu.be/V_-Qn7oEsn4

Like This I need to calibrate sir.
I'm a Rookie, and I need help to do it.

qm13, 2021-08-16:
Have you looked at the calibration sample [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/calibration).

Kanish2002, 2021-08-16:
Yes sir, I've looked at it and tried running it on visual studio 2019 but
when I open the cmakelist.txt it is opening but showing some 200 errors,
It's look like k4a library is not installed, then if I tried to install
Microsoft.Azure.kinect.sensor library from Nuget Package list, But I can't
able to manage packages in Nuget that option is disabled when I open this
calibration.
I can't able to run that calibration, that's the problem sir, So kindly
help me regarding this, I'm very eagerly waiting for your reply, I just
want to work with the camera and learn things, I know it takes time, but I
want to learn it sir.

So kindly help me out,
How can I run that calibration cpp file without errors(installing package)
successfully kindly help me out sir.

Thank you sir.