microsoft/Azure-Kinect-Sensor-SDK #1664: IR Images have invalid values after transforming

AlexCSTech, 2021-08-16:
**IR Images have invalid values after transforming**
I have been trying to transform IR images to RGB images using `depth_image_to_color_camera_custom()`, and the code works well.
But I find that the transformed IR images have many invalid values (filled by 0) because the custom transformation function relies on corresponding depth images (which also have invalid values). But the original IR images have no invalid values.
I wonder whether there is a efficient method to eliminate the invalid values (or just generate less invalid values) since I think it may just be an easy coordinate transformation.

qm13, 2021-08-16:
@AlexCSTech you are transforming the depth image (not the IR image). The depth image will have invalid values (were the depth engine could not calculate depth).

AlexCSTech, 2021-08-16:
> @AlexCSTech you are transforming the depth image (not the IR image). The depth image will have invalid values (were the depth engine could not calculate depth).

Thanks for your reply. The function `depth_image_to_color_camera_custom()` can transform depth image and custom image at the same time (it needs two paramaters `depth_image` and `custom_image`). That's to say, the custom image(which is IR image in my question) is regarded as an depth image and the process of its transformation needs the guidance of corresponding depth image pixels?