microsoft/Azure-Kinect-Sensor-SDK #1666: Azure Kinect DK cameras failed to open in the Azure Kinect Viewer?

UnityQueriesAsk, 2021-08-17:
Hi All,

I have been using the Azure Kinect DK for a long time but today all the cameras failed to open in the Azure Kinect Viewer.

I am getting the following messages pop up when connecting:

IMU failed: timed out!
IMU Data: data source failed!
Camera failed : timed out!

The Log view shows the below error :

[error] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000002 unexpected

I am using the Azure Kinect SDK v1.4.1 and the firmware version (1.6.110079014). I have tried different USB ports on my pc but still, I am getting the above errors.

Can anyone please provide a solution for these errors?

Thanks in advance.

qm13, 2021-08-19:
Please verify if the issue persists when connected to a different USB port and another PC. If it does then it is likely hardware failure.

qm13, 2021-08-19:
Also if you have a 2nd AKDK please does it exhibit the same problems on the PC?

closed 2021-10-06 by qm13

jonahpearl, 2022-02-18:
I get this problem on a home-built machine when the USB controllers aren't correct, as discussed here: https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#usb3-host-controller-compatibility. I don't really understand why some USB ports work and others don't, but it's definitely a reproducible problem.

TuttiChi, 2022-04-01:
> I get this problem on a home-built machine when the USB controllers aren't correct, as discussed here: https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#usb3-host-controller-compatibility. I don't really understand why some USB ports work and others don't, but it's definitely a reproducible problem. Fixed just by using a different USB port on the motherboard.

A ctually right. My clevo NH50 cant use this camera, but Thinkpad x390 yoga can use use it without error