microsoft/Azure-Kinect-Sensor-SDK #1667: Release Body Tracking SDK v1.1.0 NuGet package

canxerian, 2021-08-18:
**Is your feature request related to a problem? Please describe.**
The latest available version on https://www.nuget.org/packages/Microsoft.Azure.Kinect.BodyTracking/ is v1.0.1, which is incompatible with the v1.4.1 https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/

**Describe the solution you'd like**
Release Microsoft.Azure.Kinect.BodyTracking v1.1.0 as a NuGet package

skywalkerwin, 2021-08-19:
source where v1.0.1 of body tracking is incompatiable with v1.4.1?

canxerian, 2021-08-20:
Using Unity and [NugetForUnity](https://github.com/GlitchEnzo/NuGetForUnity):

Add Microsoft.Azure.Kinect.Sensor v1.4.1
Add Microsoft.Azure.Kinect.BodyTracking v1.0.1

The following error is logged to the console:
`Assembly 'Assets/Packages/Microsoft.Azure.Kinect.BodyTracking.1.0.1/lib/netstandard2.0/Microsoft.Azure.Kinect.BodyTracking.dll' will not be loaded due to errors:
Microsoft.Azure.Kinect.BodyTracking references strong named Microsoft.Azure.Kinect.Sensor Assembly references: 1.3.0.0 Found in project: 1.4.1.0.
Assembly Version Validation can be disabled in Player Settings "Assembly Version Validation"`

MFIMichael, 2022-01-24:
Is there any progress on this? I've kind of gotten to the point of thinking this is where my project might be broken, as well. 
Bizarrely, though, it does seem to work from within the Unity Editor -- but standalone builds are failing on distributed machines with the "onnxruntime.dll doesn't load correctly" error.

qm13, 2022-04-05:
Release NuGet package for BT v1.1.1

closed 2022-04-05 by qm13