microsoft/Azure-Kinect-Sensor-SDK #1669: Multi-Kinect Body Tracking SDK Error : k4abt_tracker_enqueue_capture TIME_OUT

Ideal-H, 2021-08-25:
**Describe the bug**
[Environment]
We use 5 Kinects in 1 Desktop.
We capture the frame in multi-thread.
We use 2 GPUs (2080Ti x 2).
2 Kinects are assigned to 1 GPU and the rest 3 Kinects are assigned to the other GPU.
When we used DNN model, we could not obtain the skeleton in 30 fps, so we use light model.

[Situation]
In [k4abt_tracker_enqueue_capture] function,
if we put "inf" as "wait_time" parameter, the body tracking speed is less than 30 fps and the slower the more people are.

So, we put 0 as "wait_time" parameter and could get 30 fps.
However, the problem is the body tracking SDK is sometimes malfunctioned and the return of [k4abt_tracker_enqueue_capture] function is "TIME_OUT". And never recovered unless we restart the program.

What is the problem?

**Desktop (please complete the following information):**
 - OS with Version: ubuntu18.04
 - SDK Version:  latest
 - Firmware version:  latest

kooksung930, 2021-08-25:
We found another  symptom. When the program cannot capture(?) by body tracking sdk (the call of function k4abt_tracker_enqueue_capture(tracker, capture, 0); ), single one (the sensor) cannot do that and the others are fine. I think it is caused the memory or somthing?. 

(I release the body_frame variable when the function k4abt_tracker_pop_result with wati time 0 is K4A_RESULT_SUCCEEDED)

scotthsu98052, 2021-08-31:
Please send us a dump file of the app after the error happened? We need to take a look the thread states.

Ideal-H, 2021-08-31:
Thank you for reply.
How can I check the dump file? (now I'm working with cmake in VSCODE)
And, this error is not occurred always. So, it is difficult to get some kinds of log at that time.

Now, I changed the process way of my program. 
I was use the  body tracking sdk (call the k4abt_tracker_enqueue_capture function) in multithread. But now, just capture the all data from the sensors in multithred and call the body tracking sdk function in main thread in for loop after the capturing multithread finished.
Then, the error is not appeared until now and the frame drop is decreased.

According this trial, I think calling the body tracking sdk in multithread is not safe. (maybe the cause of memory of GPU)
Am I right?

Thanks.

Ideal-H, 2021-09-06:
Sorry for late.

This is the thread stack

Thread 164 (Thread 0x7ffd1e7f4700 (LWP 3925)):
#0  0x00007ffff6fcead3 in futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x5555aa9022e8) at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  0x00007ffff6fcead3 in __pthread_cond_wait_common (abstime=0x0, mutex=0x5555aa9022f0, cond=0x5555aa9022c0) at pthread_cond_wait.c:502
#2  0x00007ffff6fcead3 in __pthread_cond_wait (cond=0x5555aa9022c0, mutex=0x5555aa9022f0) at pthread_cond_wait.c:655
#3  0x00007fffedcbe8bc in std::condition_variable::wait(std::unique_lock<std::mutex>&) () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff6d2aaf9 in DataConsumerThread<Useful::Buffer<DnnPredictionBlob> >::GetNextData(Useful::Buffer<DnnPredictionBlob>&) (data=..., this=<optimized out>) at /usr/include/c++/8/bits/stl_deque.h:1367
#5  0x00007ffff6d2aaf9 in DataConsumerThread<Useful::Buffer<DnnPredictionBlob> >::ThreadEntry() (this=0x5555aa902248) at ../src/ThreadLib/DataConsumerThread.h:106
#6  0x00007ffff6d29e42 in ProcessingThread::Start()::{lambda()#1}::operator()() const (__closure=<optimized out>) at ../src/ThreadLib/ProcessingThread.h:19
#7  0x00007ffff6d29e42 in std::__invoke_impl<void, ProcessingThread::Start()::{lambda()#1}>(std::__invoke_other, ProcessingThread::Start()::{lambda()#1}&&) (__f=...) at /usr/include/c++/8/bits/invoke.h:60
#8  0x00007ffff6d29e42 in std::__invoke<ProcessingThread::Start()::{lambda()#1}>(std::__invoke_result&&, (ProcessingThread::Start()::{lambda()#1}&&)...) (__fn=...) at /usr/include/c++/8/bits/invoke.h:95
#9  0x00007ffff6d29e42 in std::thread::_Invoker<std::tuple<ProcessingThread::Start()::{lambda()#1}> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) (this=<optimized out>) at /usr/include/c++/8/thread:244
#10 0x00007ffff6d29e42 in std::thread::_Invoker<std::tuple<ProcessingThread::Start()::{lambda()#1}> >::operator()() (this=<optimized out>) at /usr/include/c++/8/thread:253
#11 0x00007ffff6d29e42 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<ProcessingThread::Start()::{lambda()#1}> > >::_M_run() (this=0x5555b040f730) at /usr/include/c++/8/thread:196
#12 0x00007fffedcc46df in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#13 0x00007ffff6fc86db in start_thread (arg=0x7ffd1e7f4700) at pthread_create.c:463
#14 0x00007fffed71f71f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 163 (Thread 0x7ffd1eff5700 (LWP 3924)):
#0  0x00007ffff6fcead3 in futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x5555aa9020b4) at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  0x00007ffff6fcead3 in __pthread_cond_wait_common (abstime=0x0, mutex=0x5555aa9020b8, cond=0x5555aa902088) at pthread_cond_wait.c:502
#2  0x00007ffff6fcead3 in __pthread_cond_wait (cond=0x5555aa902088, mutex=0x5555aa9020b8) at pthread_cond_wait.c:655
#3  0x00007fffedcbe8bc in std::condition_variable::wait(std::unique_lock<std::mutex>&) () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff6d2ad19 in DataConsumerThread<Useful::Buffer<SensorCaptureData> >::GetNextData(Useful::Buffer<SensorCaptureData>&) (data=..., this=<optimized out>) at /usr/include/c++/8/bits/stl_deque.h:1367
#5  0x00007ffff6d2ad19 in DataConsumerThread<Useful::Buffer<SensorCaptureData> >::ThreadEntry() (this=0x5555aa902010) at ../src/ThreadLib/DataConsumerThread.h:106
#6  0x00007ffff6d29e42 in ProcessingThread::Start()::{lambda()#1}::operator()() const (__closure=<optimized out>) at ../src/ThreadLib/ProcessingThread.h:19
#7  0x00007ffff6d29e42 in std::__invoke_impl<void, ProcessingThread::Start()::{lambda()#1}>(std::__invoke_other, ProcessingThread::Start()::{lambda()#1}&&) (__f=...) at /usr/include/c++/8/bits/invoke.h:60
#8  0x00007ffff6d29e42 in std::__invoke<ProcessingThread::Start()::{lambda()#1}>(std::__invoke_result&&, (ProcessingThread::Start()::{lambda()#1}&&)...) (__fn=...) at /usr/include/c++/8/bits/invoke.h:95
#9  0x00007ffff6d29e42 in std::thread::_Invoker<std::tuple<ProcessingThread::Start()::{lambda()#1}> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) (this=<optimized out>) at /usr/include/c++/8/thread:244
#10 0x00007ffff6d29e42 in std::thread::_Invoker<std::tuple<ProcessingThread::Start()::{lambda()#1}> >::operator()() (this=<optimized out>) at /usr/include/c++/8/thread:253
#11 0x00007ffff6d29e42 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<ProcessingThread::Start()::{lambda()#1}> > >::_M_run() (this=0x5555b040f750) at /usr/include/c++/8/thread:196
#12 0x00007fffedcc46df in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#13 0x00007ffff6fc86db in start_thread (arg=0x7ffd1eff5700) at pthread_create.c:463
#14 0x00007fffed71f71f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 162 (Thread 0x7ffd1f7f6700 (LWP 3923)):
#0  0x00007fffed719639 in syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
#1  0x00007fffe5259bac in nsync::nsync_mu_semaphore_p_with_deadline(nsync::nsync_semaphore_s_*, timespec) () at /usr/lib/libonnxruntime.so.1.6.0
#2  0x00007fffe5259931 in nsync::nsync_sem_wait_with_cancel_(nsync::waiter*, timespec, nsync::nsync_note_s_*) () at /usr/lib/libonnxruntime.so.1.6.0
#3  0x00007fffe5258878 in nsync::nsync_cv_wait_with_deadline_generic(nsync::nsync_cv_s_*, void*, void (*)(void*), void (*)(void*), timespec, nsync::nsync_note_s_*) () at /usr/lib/libonnxruntime.so.1.6.0
[... cut ...]
#6  0x00007fffed71f71f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 1 (Thread 0x7ffff7f3d000 (LWP 3346)):
#0  0x00007ffff6fc9d2d in __GI___pthread_timedjoin_ex (threadid=140723210819328, thread_return=0x0, abstime=0x0, block=<optimized out>) at pthread_join_common.c:89
#1  0x00007fffedcc4933 in std::thread::join() () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x0000555555590d17 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /home/sa/SA/New_GIST_PRO/N_GIST_PRO_ADHD/Tracking_system_ADHD/main.cpp:1920

Thank you.

Ideal-H, 2021-09-08:
Thank you !!, and I sent you e-mail <scotthsu@microsoft.com>

closed 2021-12-10 by qm13