microsoft/Azure-Kinect-Sensor-SDK #1671: Depth sensor can get unresponsive and disapear from the USB bus on driver crash

jmachowinski, 2021-08-30:
**Describe the bug**
The depth sensor (045e:097c) can get into a state, were it won't show up on the
usb bus any more. The only recovery from this state, is to cut the power
to the sensor.

**To Reproduce**
1. Start the driver, and start streaming
    (We used the ros driver for this)
2. Kill the driver process
3. Check if the device is still present on the bus
4. If still present, repeat from step 1. 
Note, it takes around 100 restarts, until this happens.

**Expected behavior**
Sensor should be responsive.

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04 / Ubuntu 20.04
 - SDK Version:  1.4.1
 - Firmware version:   RGB 1.6.110, Depth 1.6.79

qm13, 2021-08-30:
Please test with the beta firmware [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/firmware-beta).

jmachowinski, 2021-08-31:
We'll start a new stress test with the beta firmware.

jmachowinski, 2021-09-01:
Seems to be fixed, 1280 driver restarts, camera still present, we'll leave it running over night...

jmachowinski, 2021-09-06:
We had it running for a few days, 4798 restart without this device disappearing. 
Is this firmware safe for productive environments, or does is contain other regressions ?

qm13, 2021-10-06:
The firmware is safe to use. It will be officially release as part of Sensor SDK v1.5.

closed 2021-10-06 by qm13

ekdrnqkdrn, 2021-10-06:
Hello.
I had this problem and updated the firmware to Azure Kinect DK _Fw_1.6.110080014B.bin, but the same problem occurs when the PC reboots.
[image]