microsoft/Azure-Kinect-Sensor-SDK #1672: Body tracking sdk captured frame (color and depth image) is not same with the captured from sensor sdk

Ideal-H, 2021-09-06:
Hello, I'm trying to use multiple kinect sensors in single PC and Body tracking SDK.
Until now, I'm capturing the frame using [k4a_device_get_capture] function.
But, through some test, The frame (the color and depth image) captured from [k4a_device_get_capture] function and the frame that release the frame captured from [k4a_device_get_capture] and get re-capture(? this function may just get the older frame) using [k4abt_frame_get_capture] function is different.
It seems the frame from [k4a_device_get_capture] function is not latest frame (I print the 2D skeleton points in both frames and checked).

is there no problem?

I'm using [device_config.synchronized_images_only = true] parameter and there is no temporal smoothing in body tracking sdk (    k4abt_tracker_set_temporal_smoothing function is not used)

please help.
Thank you

qm13, 2021-09-08:
@Ideal-H there seems to be two questions in this issue - 1) regarding accessing the RGB image associated with the pose estimation and 2) controlling temporal smoothing.

I am a little confused by the first one. It sounds like you are using k4a_device_get_capture to get the next capture in the AKDK stream and passing that to body tracking and then using k4abt_frame_get_capture to get the RGB image associated with the depth data used to generate the pose estimation. Correct? What is the issue that you are seeing?

Regarding temporal smoothing are you stating that you see no difference in the pose estimations in a movement sequence i.e. this factor impacts pose estimation over a sequence of captures?

Ideal-H, 2021-09-08:
Sorry for confusing you. 
the capturing code in my program is like 

    capture = nullptr;
    k4a_wait_result_t getCaptureResult = k4a_device_get_capture(device, &capture, -1);                        ---- first capture
    color_image = k4a_capture_get_color_image(originalCapture);
    depth_image = k4a_capture_get_depth_image(originalCapture);

    if (getCaptureResult == K4A_WAIT_RESULT_SUCCEEDED)
    {
        k4a_wait_result_t queueCaptureResult = k4abt_tracker_enqueue_capture(tracker, capture, 0);
        k4a_capture_release(capture);                                                                                                           ---- release the capture

        if (queueCaptureResult == K4A_WAIT_RESULT_FAILED)
            return;

        body_frame = nullptr;
        k4a_wait_result_t popFrameResult = k4abt_tracker_pop_result(tracker, &body_frame, 0);
        if (popFrameResult == K4A_WAIT_RESULT_SUCCEEDED)
        {
            k4a_capture_t originalCapture = k4abt_frame_get_capture(body_frame);                                  ----- get latest capture
            color_image = k4a_capture_get_color_image(originalCapture);
            depth_image = k4a_capture_get_depth_image(originalCapture);

my question is that
the two image data(color_iamge, depth_image) in [first capture] and [get latest capture] is different. The first capture image is not latest.

For the temporal smoothing, I just want to notice that I don't use it

Thanks