microsoft/Azure-Kinect-Sensor-SDK #1675: How to get (Euclidian) distance data

MseoKang, 2021-09-08:
Is there a way to get the raw distance from the Azure Kinect to each point? According to the [documentation,](https://docs.microsoft.com/en-us/azure/kinect-dk/depth-camera) the depth values are not the distances from each point to the Azure Kinect, but the z-coordinates. I was wondering if there was a way to measure the Euclidian distance from the Azure Kinect's origin to an arbitrary point.

qm13, 2021-09-13:
The depth value is the distance in mm of the pixel from the depth camera.

MseoKang, 2021-09-14:
> 
> 
> The depth value is the distance in mm of the pixel from the depth camera.

I tested it, and the depth value was not the distance between the object and the depth camera, but the z-coordinates. For example, two objects at (-300, 0, 600) and (600, 0, 600) gave the same depth values, even though their distances were different. I know that the depth camera uses a time-of-flight sensor, so the device should know the raw distance somewhere.

hlyf-xs, 2022-01-03:
I also test the transform depth image to color image, but the depth value from transformed depth image is not the distance, I don't know how to solve the problem.