microsoft/Azure-Kinect-Sensor-SDK #1676: The saved data of the mic array is not sync'ed with that of the color camera.

TristanWYL, 2021-09-08:
**Describe the bug**

Our case needs to save the data from the mic array and the color camera simultaneously. We use [this](https://github.com/TristanWYL/Azure-Kinect-Recorder) to do the work. However, an issue is that the saved audio is not sync'ed strictly with the video. 

For example, given a recording of 1.5 hours, the video is always a little shorter by around 2 seconds than the audio.

To fix this, we adapt the timestamp of each image frame as the timestamp when the host receives the frame. Doing this could make the length of the audio and video consistent. However, doing this will make the frame unevenly distributed along the timeline, as different frames may take different time to arrive at the host.

**To Reproduce**
You could conduct a test with  [this](https://github.com/TristanWYL/Azure-Kinect-Recorder).

**Expected behavior**
As the recording of audio and video start/end at the same time, the resulted audio and video should have the same duration.

**Desktop (please complete the following information):**
 - Windows 10 Home

**My findings so far**
For your reference, the video generation and the audio generation may depend on different clock. Probably, the audio uses the host's closk, while the video uses the Kinect's built-in clock. In this case, the resulted audio and video may have different duration.

TristanWYL, 2021-09-09:
Please help answer, as it is quite URGENT!

qm13, 2021-09-13:
There is no synchronization between the mic array and color camera.

TristanWYL, 2021-09-14:
> There is no synchronization between the mic array and color camera.

Thanks for your reply. So may I infer that the generation of audio samples and video frames is based on different clocks inside the device?

qm13, 2021-09-16:
Yes, the clocks are different.