microsoft/Azure-Kinect-Sensor-SDK #1682: Unity Body Tracker cannot start in GPU mode

adeckerroto, 2021-09-23:
**Describe the bug**
Unity cannot start running body tracker in Unity when the TrackerConfiguration.ProcessingMode is set to TrackerProcessingMode.Gpu . This can be done with other projects, but can most quickly be seen in the associated [Azure Kinect Samples](https://github.com/microsoft/Azure-Kinect-Samples). Reproduction steps are with this project for easy repro.

The tracker does work with the modes TrackerProcessingMode.Cuda, and TrackerProcessingMode.Cpu. Although those modes have a significant latency. Cuda with about .4 seconds and Cpu with nearly a whole second of latency. This latency does not show up when using the Azure Kinect Body Tracking Viewer standalone application, only when the data transfers into the Unity editor. In the structure for TrackerConfirguration it is mentioned that Gpu is the most performant and would be great to lower the latency, through even at high latency it is still really accurate.

NOTE: I am aware of issues #1147 and #1367 as well as of pull request #50. None of these solutions have fixed my problem and allowed me to run the application in Gpu mode.

**To Reproduce**
1. Follow the [Unity Sample Setup Instructions](https://github.com/microsoft/Azure-Kinect-Samples/tree/2a1e75a0ea7a108415b39f9df66cfe09f856029d/body-tracking-samples/sample_unity_bodytracking).
2. Add the required dependencies manually as the MoveLibraryFiles.bat is still out of date.
3. Ensure that in SkeletalTrackingProvider, line 46uses TrackerProcessingMode.Gpu as seen below.
`using (Tracker tracker = Tracker.Create(deviceCalibration, new TrackerConfiguration() { ProcessingMode = TrackerProcessingMode.Gpu, SensorOrientation = SensorOrientation.Default }))`
4. Observe Error Behavior

**Observed Error**
The application fails to begin tracking. The following error is thrown AzureKinectBodyTrackingCreateException: result = K4A_RESULT_FAILED. This occurs on the execution of the following line:
`using (Tracker tracker = Tracker.Create(deviceCalibration, new TrackerConfiguration() { ProcessingMode = TrackerProcessingMode.Gpu, SensorOrientation = SensorOrientation.Default }))`

**Expected behavior**
Tracking starts in Gpu more without crashing and gives low latency, accurate body tracking in Unity.

**Desktop (please complete the following information):**
 - OS with Version: Windows 10.0.19043
 - SDK Version:  1.1.0
 - Firmware version: factory default

mara90, 2021-11-15:
I'm facing the same issue as @adeckerroto, I'm using TrackerProcessingMode.Cuda in my project and the delay is very noticeable. 
Are there any updates or fixes on this issue?

crawfis, 2022-03-31:
Same problem running on an Desktop. Works in CPU mode (very slow) but produces an error in GPU mode. 
Intel i7-9700K
nVidia GeForce RTX 2070 SUPER
SDK 1.1.1
Windows 10 (up to date)

crawfis, 2022-03-31:
FYI, the camera does open.
1) I tried all combinations of GPU, CPU, Cuda and DirectML and also tried the lite ProcessModel. Only Cpu worked.
2) Trying to expand the device info in Visual Studio causes it to hang (consistenly), but killing Unity brings it back. It is stopped on a breakpoint.

Plymouth-IML, 2022-05-18:
Having the same issue works with TrackerProcessingMode.Cpu but not TrackerProcessingMode.Cuda

Windows 10 : RTX 2080

Plymouth-IML, 2022-05-24:
Not sure if this thread is still being monitored but few more details below : 

Only works with TrackerProcessingMode.Cpu but not TrackerProcessingMode.Cuda or TrackerProcessingMode.Gpu, however the stand alone app : Azure Kinect Body Tracking Viewer works with k4abt_simple_3d_viewer.exe CUDA 

Have tried rolling back to old versions of SDKs, CUDA, Unity with no change of results checked all require files are part of Unity project in correct locations. Also tried adding C:\Program Files\Azure Kinect Body Tracking SDK\tools to Environmental Variables and running the Unreal project file with not success. 

Azure Kinect Body Tracking SDK 1.1.1
Azure Kinect SDK v1.4.1
(Installed to default locations)

Processor	Intel(R) Core(TM) i7-9700 CPU @ 3.00GHz   3.00 GHz
Installed RAM	64.0 GB (63.8 GB usable)
System type	64-bit operating system, x64-based processor

Edition	Windows 10 Enterprise
Version	21H2
OS build	19044.1706

NVIDIA GeForce RTX 2080
Driver version 516.01
CUDA 11.7.0_516.01

fkavache, 2022-06-03:
Hi @crawfis, @Plymouth-IML

Check the latest changes in the unity samples project.
Please follow the updated instructions.

Plymouth-IML, 2022-06-06:
> Hi @crawfis, @Plymouth-IML
> 
> Check the latest changes in the unity samples project. Please follow the updated instructions.

Many thanks fkavache this is now working! One note for those installing the MoveLibraryFile.bat file did not copy the below files so have to do them manually : 
directml.dll
onnxruntime.dll
onnxruntime_providers_cuda.dll
onnxruntime_providers_shared.dll
onnxruntime_providers_tensorrt.dll

closed 2022-06-09 by qm13