microsoft/Azure-Kinect-Sensor-SDK #1686: An error occurs while synchronizing multiple kinects to get a color image and a depth image

hirokim, 2021-10-05:
**Describe the bug**

Color image and Depth image are acquired from each kinect by synchronizing 4 kinects in a Daisy-chain configuration.
It works fine for the first few minutes to tens of minutes, but after a while it quits with an error.
Do you know what caused it?
It would be very appreciate if you could advise me what to do to make it work properly.

**Desktop**
 - OS with Version: Windows 10 Pro 1909
 - SDK Version:  1.4.1
 - Firmware version:  1.6.110079014

**Logs**
```
New logging session started

[2021-10-05 16:11:04.607] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (343): find_libusb_device(). Container ID found: {d77fecb9-4e00-4376-63b8-9ea830be29ac} 
[2021-10-05 16:11:04.607] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (224): populate_serialnumber(). Serial Number found 001069501412
[2021-10-05 16:11:04.635] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (343): find_libusb_device(). Container ID found: {d77fecb9-4e00-4376-63b8-9ea830be29ac} 
[2021-10-05 16:11:04.635] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (224): populate_serialnumber(). Serial Number found 001069501412
[2021-10-05 16:11:04.663] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (225): ******************** Device Info ********************
[2021-10-05 16:11:04.663] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (226): K4A SDK version:     1.4.1
[2021-10-05 16:11:04.689] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (232): Serial Number:       001069501412
[2021-10-05 16:11:04.689] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (236): RGB Sensor Version:  1.6.110
[2021-10-05 16:11:04.689] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (239): Depth Sensor Version:1.6.79
[2021-10-05 16:11:04.689] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (242): Mic Array Version:   1.6.14
[2021-10-05 16:11:04.689] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (245): Sensor Config:       6109.7
[2021-10-05 16:11:04.689] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (248): Build type:          Release
[2021-10-05 16:11:04.689] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (253): Signature type:      MSFT
[2021-10-05 16:11:04.689] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (259): ****************************************************
[2021-10-05 16:11:04.813] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (343): find_libusb_device(). Container ID found: {0ab8e449-4fdb-4ee9-67ae-cdc79f4989e8} 
[2021-10-05 16:11:04.814] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (224): populate_serialnumber(). Serial Number found 001097401412
[2021-10-05 16:11:04.840] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (343): find_libusb_device(). Container ID found: {0ab8e449-4fdb-4ee9-67ae-cdc79f4989e8} 
[2021-10-05 16:11:04.841] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (224): populate_serialnumber(). Serial Number found 001097401412
[2021-10-05 16:11:04.868] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (225): ******************** Device Info ********************
[2021-10-05 16:11:04.868] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (226): K4A SDK version:     1.4.1
[2021-10-05 16:11:04.894] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (232): Serial Number:       001097401412
[2021-10-05 16:11:04.894] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (236): RGB Sensor Version:  1.6.110
[2021-10-05 16:11:04.894] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (239): Depth Sensor Version:1.6.79
[2021-10-05 16:11:04.894] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (242): Mic Array Version:   1.6.14
[2021-10-05 16:11:04.894] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (245): Sensor Config:       6109.7
[2021-10-05 16:11:04.894] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (248): Build type:          Release
[2021-10-05 16:11:04.894] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (253): Signature type:      MSFT
[2021-10-05 16:11:04.894] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (259): ****************************************************
[2021-10-05 16:11:04.950] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (343): find_libusb_device(). Container ID found: {5caf2816-52c1-4626-8381-0ee254592428} 
[2021-10-05 16:11:04.951] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (224): populate_serialnumber(). Serial Number found 000327294412
[2021-10-05 16:11:04.976] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (343): find_libusb_device(). Container ID found: {5caf2816-52c1-4626-8381-0ee254592428} 
[2021-10-05 16:11:04.977] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (224): populate_serialnumber(). Serial Number found 000327294412
[2021-10-05 16:11:05.005] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (225): ******************** Device Info ********************
[2021-10-05 16:11:05.005] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (226): K4A SDK version:     1.4.1
[2021-10-05 16:11:05.030] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (232): Serial Number:       000327294412
[2021-10-05 16:11:05.030] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (236): RGB Sensor Version:  1.6.110
[2021-10-05 16:11:05.030] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (239): Depth Sensor Version:1.6.79
[2021-10-05 16:11:05.030] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (242): Mic Array Version:   1.6.14
[2021-10-05 16:11:05.030] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (245): Sensor Config:       6109.7
[2021-10-05 16:11:05.030] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (248): Build type:          Release
[2021-10-05 16:11:05.030] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (253): Signature type:      MSFT
[2021-10-05 16:11:05.030] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (259): ****************************************************
[2021-10-05 16:11:05.097] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (343): find_libusb_device(). Container ID found: {0d52bca0-51e1-4abc-c6aa-6d072a916e58} 
[2021-10-05 16:11:05.098] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (224): populate_serialnumber(). Serial Number found 000267694612
[2021-10-05 16:11:05.144] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (343): find_libusb_device(). Container ID found: {0d52bca0-51e1-4abc-c6aa-6d072a916e58} 
[2021-10-05 16:11:05.145] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (224): populate_serialnumber(). Serial Number found 000267694612
[2021-10-05 16:11:05.173] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (225): ******************** Device Info ********************
[2021-10-05 16:11:05.173] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (226): K4A SDK version:     1.4.1
[2021-10-05 16:11:05.198] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (232): Serial Number:       000267694612
[2021-10-05 16:11:05.198] [critical] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (236): RGB Sensor Version:  1.6.110
[... cut ...]
[2021-10-05 16:50:18.620] [info] [t=26068] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\queue\queue.c (300): queue_stop(). Queue "Queue_depth" stopped, shutting down and notifying consumers.
[2021-10-05 16:50:18.620] [info] [t=26068] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\queue\queue.c (300): queue_stop(). Queue "Queue_color" stopped, shutting down and notifying consumers.
[2021-10-05 16:50:18.646] [info] [t=35572] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (952): k4a_device_stop_cameras(). k4a_device_stop_cameras stopped

```

yuzhipeng66, 2022-11-09:
Hello, I meet the same problem. Is this problem solved?

hirokim, 2022-11-09:
@yuzhipeng66 
Hello, I couldn't solve it.
I created a process monitoring application to restart the Kinect application when it crashes.