microsoft/Azure-Kinect-Sensor-SDK #1697: How do I get MJPG images？C++

Havehandssook, 2021-10-30:
When I use the following code I can get color images in BGRA format.

    k4a_device_t device = NULL; 
	K4A_FAILED(k4a_device_open(0, &device));

	k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
	config.camera_fps = K4A_FRAMES_PER_SECOND_30;
	config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
	config.color_resolution = K4A_COLOR_RESOLUTION_720P;

	k4a_device_start_cameras(device, &config); 

		k4a_capture_t sensor_capture;
		k4a_wait_result_t get_capture_result = k4a_device_get_capture(device, &sensor_capture, K4A_WAIT_INFINITE);
		k4a_image_t rbgimage = k4a_capture_get_color_image(sensor_capture); 
		Mat color_image = Mat(k4a_image_get_height_pixels(rbgimage), 
                                  k4a_image_get_width_pixels(rbgimage), 
                                  CV_8UC4,
                                  k4a_image_get_buffer(rbgimage));
		
		imshow("color", color_image); 

How do I get MJPG images？Or is YUY2? Or is NV12?

I tried to change the K4A_IMAGE_FORMAT_COLOR_BGRA32 to the K4A_IMAGE_FORMAT_COLOR_MJPG，i get an error.
Is there anything else that needs to be changed?
Hope to get your help, thank you!!!

Havehandssook, 2021-11-02:
[image]
I found this in the document, but I don't know how to use JPEG decoder... ...
  @qm13

UnaNancyOwen, 2021-11-17:
If you have set image format to `K4A_IMAGE_FORMAT_COLOR_MJPG`, the buffer that retrieved by `k4a_image_get_buffer()` is jpeg encoded data. You need to decode it yourself.

```cpp
// get color image handle
k4a_image_t image = k4a_capture_get_color_image( capture );

// get color image buffer and size
uint8_t* buffer = k4a_image_get_buffer( image );
size_t size = k4a_image_get_size( buffer );

// decode motion jpeg using opencv
std::vector<uint8_t> jpeg( buffer, buffer + size );
cv::Mat mat = cv::imdecode( jpeg, cv::IMREAD_ANYCOLOR );

// show image
cv::imshow( "motion jpeg", mat );
```

If you want to convert `k4a_image_t` to `cv::Mat` easily, you can use `k4a_get_mat()`. It is support all image format. 
https://github.com/UnaNancyOwen/AzureKinectSample/blob/master/sample/c/color/util.h

```cpp
#include "util.h"
```
```cpp
// get color image handle
k4a_image_t image = k4a_capture_get_color_image( capture );

// get opencv mat from image handle
cv::Mat mat = k4a_get_mat( image );
```

Havehandssook, 2021-11-17:
> If you have set image format to `K4A_IMAGE_FORMAT_COLOR_MJPG`, the buffer that retrieved by `k4a_image_get_buffer()` is jpeg encoded data. You need to decode it yourself.
> 
> ```c++
> // get color image handle
> k4a_image_t image = k4a_capture_get_color_image( capture );
> 
> // get color image buffer and size
> uint8_t* buffer = k4a_image_get_buffer( image );
> size_t size = k4a_image_get_size( buffer );
> 
> // decode motion jpeg using opencv
> std::vector<uint8_t> jpeg( buffer, buffer + size );
> cv::Mat mat = cv::imdecode( jpeg, cv::IMREAD_ANYCOLOR );
> 
> // show image
> cv::imshow( "motion jpeg", mat );
> ```
> 
> If you want to convert `k4a_image_t` to `cv::Mat` easily, you can use `k4a_get_mat()`. It is support all image format. https://github.com/UnaNancyOwen/AzureKinectSample/blob/master/sample/c/color/util.h
> 
> ```c++
> #include "util.h"
> ```
> 
> ```c++
> // get color image handle
> k4a_image_t image = k4a_capture_get_color_image( capture );
> 
> // get opencv mat from image handle
> cv::Mat mat = k4a_get_mat( image );
> ```

So many days have passed, I am so surprised to get your help, I will try the method you provided as soon as possible, at the same time, I saw that your project is also for the official API class encapsulation, I am doing the same thing, I think I will learn a lot from your project, thank you again for your reply!

Havehandssook, 2021-11-17:
> If you have set image format to `K4A_IMAGE_FORMAT_COLOR_MJPG`, the buffer that retrieved by `k4a_image_get_buffer()` is jpeg encoded data. You need to decode it yourself.
>
> ```c++
> // get color image [...]

        int main()
        {
	          k4a_device_t device = NULL; 
	          K4A_FAILED(k4a_device_open(0, &device));
          
	          k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
	          config.camera_fps = K4A_FRAMES_PER_SECOND_30;
	          config.color_format = K4A_IMAGE_FORMAT_COLOR_MJPG;
	          config.color_resolution = K4A_COLOR_RESOLUTION_720P;
	          config.depth_delay_off_color_usec = false;
          
	          k4a_device_start_cameras(device, &config); 
          
	          while (true) 
	          {
		          k4a_capture_t sensor_capture;
		          k4a_wait_result_t get_capture_result = k4a_device_get_capture(device, &sensor_capture, K4A_WAIT_INFINITE);
		          k4a_image_t rbgimage = k4a_capture_get_color_image(sensor_capture); 
		          //Mat color_image = Mat(k4a_image_get_height_pixels(rbgimage), k4a_image_get_width_pixels(rbgimage), CV_8UC4, k4a_image_get_buffer(rbgimage));
		          
		          cv::Mat color_image = k4a_get_mat(rbgimage);
          
		          imshow("color", color_image); 
		          cout << "1" << endl;
          
		          k4a_image_release(rbgimage); 
		          waitKey(1);
	          }
          
	          k4a_device_stop_cameras(device); 
	          k4a_device_close(device);
	          waitKey(0);
	          return 0;
        }

I tried following your instructions, but the program stalled after only 10 frames.I added output 1
[image]
 to the window.

UnaNancyOwen, 2021-11-17:
You have not released the capture handle.
Please release capture handle using `k4a_capture_release()` in proper line of your program.
Usually, that is do after getting the k4a_image_t.

````cpp
// get capture
k4a_capture_t capture;
k4a_wait_result_t result = k4a_device_get_capture( device, &capture, K4A_WAIT_INFINITE );

// get image
k4a_image_t image = k4a_capture_get_color_image( capture );

// release capture
k4a_capture_release( capture ); /* this */

// get mat
cv::Mat mat = k4a_get_mat( image );

// release image
k4a_image_release( image );
````

Havehandssook, 2021-11-17:
> You have not released the capture handle. Please release capture handle using `k4a_capture_release()` in proper line of your program. Usually, that is do after getting the k4a_image_t.
> 
> ```c++
> // get capture
> k4a_capture_t capture;
> k4a_wait_result_t result = k4a_device_get_capture( device, &capture, K4A_WAIT_INFINITE );
> 
> // get image
> k4a_image_t image = k4a_capture_get_color_image( capture );
> 
> // release capture
> k4a_capture_release( capture ); /* this */
> 
> // get mat
> cv::Mat mat = k4a_get_mat( image );
> 
> // release image
> k4a_image_release( image );
> ```

I also noticed this just now, it is running normally, thank you!May I ask you another question by the way?# 1695

Havehandssook, 2021-11-17:
> You have not released the capture handle. Please release capture handle using `k4a_capture_release()` in proper line of your program. Usually, that is do after getting the k4a_image_t.
>
> ```c++
> // [...]

I have ported these apis using routines to correct the original depth map, but I want to correct the depth map after the depth map is aligned with the color map. These apis are no longer useful. What should I do?
[image]

UnaNancyOwen, 2021-11-17:
If this problem has been solved, you should close this issue.
And, you should not talk about other issues in here.

Havehandssook, 2021-11-17:
> If this problem has been solved, you should close this issue. And, you should not talk about other issues in here.

I'm very sorry, this problem has been solved!I will close this problem immediately.

closed 2021-11-17 by Havehandssook

wangcc123-dave, 2022-11-01:
> 

Thank you very much.The k4a_get_mat function helps me a lot.

wangcc123-dave, 2022-11-01:
> If you have set image format to `K4A_IMAGE_FORMAT_COLOR_MJPG`, the buffer that retrieved by `k4a_image_get_buffer()` is jpeg encoded data. You need to decode it yourself.
>
> ```c++
> // get color image [...]

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1697#issuecomment-970901949

wangcc123-dave, 2022-11-01:
> If you have set image format to `K4A_IMAGE_FORMAT_COLOR_MJPG`, the buffer that retrieved by `k4a_image_get_buffer()` is jpeg encoded data. You need to decode it yourself.
>
> ```c++
> // get color image [...]

Thank you!The k4a_get_mat( ) function helps me a lot！