microsoft/Azure-Kinect-Sensor-SDK #1698: Multiple Sensors Order?

arkader, 2021-10-30:
Hi Everyone, 

I've got 6 Azure Kinects sensors connected my computer (using PCI Express Card USB port mainly).
And I've noticed that the sensors are not listed in the order that I've plugged them. 
Do you know what's the sensor order logic? 
I've got my 6 camera configured in a daisy chain configuration, but the sensor numbers are not aligned, which makes it confusing. 
I've tried pluging them again in different order, but no difference.

This is the order I have
```
Sensor 1 is camera 1 - 0009.4531.271
Sensor 2 is camera 5 - 0007.7861.2712
Sensor 3 is camera 3 - 0009.4711.2712
Sensor 4 is camera 2 - 0009.4961.2712
Sensor 5 is camera 6 - 0009.5771.2712
Sensor 6 is camera 4 - 0007.3561.2712
```
Thank you for your help

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  
```
 RGB camera firmware:      1.6.110
Depth camera firmware:    1.6.79
Depth config file:        6109.7
Audio firmware:           1.6.14
Build Config:             Production
Certificate Type:         Microsoft
```

qm13, 2021-11-02:
@arkader the order is determined by Windows the first time you plug the devices into the PC. The order is then preserved.

arkader, 2021-11-03:
Thank you @qm13

closed 2021-11-03 by arkader

AstitvaSri, 2021-12-02:
> @arkader the order is determined by Windows the first time you plug the devices into the PC. The order is then preserved.
@qm13 Is there any way to reset this order in Ubuntu?