microsoft/Azure-Kinect-Sensor-SDK #1699: Azure Kinect Camera not detected in Viewer - Windows 11

darthcodar, 2021-10-30:
Device is connected via USB 3.0 and LED on back is solid white. 
Azure Kinect Viewer fails to detect the device.

Desktop:

    Display Drivers: latest nVidia GeForce
    OS with Version: Windows 11 Pro, version 21H2
    SDK Version: 1.4.1
    Firmware version: (unable to connect)

Additional context
Unable to Find device in Azure Kinect Viewer app, tried refresh devices, reboot.

I can run the Kinect as a webcam/camera through Windows Camera app. 
Usual Webcam has been disconnected
Tried on work laptop, same issue - is this a Windows 11 incompatibility?

Azure Kinect 4K Camera is detected in the Device Manager under cameras
Azure Kinect 4K Camera and Azure Kinect Depth Camera are detected in the Device Manager under USB devices

closed 2021-11-01 by qm13

pswot, 2022-01-28:
Have this bug been solved?

darthcodar, 2022-01-31:
Solved with update to newest firmware. Thanks a lot