microsoft/Azure-Kinect-Sensor-SDK #1700: Latest firmware versioning issue on SDK v1.4.1?

arkader, 2021-10-31:
Hi Guys, 

I'm using 6 Azure Kinects. 
I've done a factory reset for all of them, and what I've noticed is that the factory firmware is higher than the firmware provided by the latest sdk v1.4.1, is that just a versioning typo issue? Or should I leave the factory firmware and not use the one from the SDK.

Factory Reset Firmware
[image]

Firmware after updating manually:
`RGB camera firmware:  1.6.110
  Depth camera firmware: 1.6.79
  Depth config file: 6109.7
  Audio firmware: 1.6.14
  Build Config: Production
  Certificate Type:  Microsoft`
  
  You'll notice that the `Depth camera firmware` is downgrading after the update

**Desktop (please complete the following information):**
 - OS with Version:  Windows 10 Pro
 - SDK Version: v1.4.1v
 - Firmware version:  AzureKinectDK_Fw_1.6.110079014

Thank you for your help

qm13, 2021-11-02:
@arkader the factory has started loading the latest firmware. The v1.5 SDK which carries the new firmware is delayed. You can also find the firmware here https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/firmware

arkader, 2021-11-02:
Thank you for this information @qm13 
I'll upgrade all the sensors to this latest version then:
AzureKinectDK_Fw_1.6.110080014.bin

closed 2021-11-03 by arkader