microsoft/Azure-Kinect-Sensor-SDK #1704: Body tracking k4abt_simple_3d_viewer crashes on start up

pucciland95, 2021-11-06:
Hello everybody, I am trying to set up the body tracking abilities that the Azure Kinect is supposed to have. Hereafter my PC configuration and the drivers installed:

OS: Ubuntu 20.04
GPU: Gtx 1050
GPU driver and Cuda version (output of nvidia-smi): NVIDIA-SMI 495.29.05    Driver Version: 495.29.05    CUDA Version: 11.5

Camera settings
 - SDK Version: 1.4
 - Body tracking Version: 1.1

**Note** at this time the Cuda version was cuda10.x
I have installed the latest body tracking packages:

- libk4abt1.1_1.1.0_amd64.deb
- libk4abt1.1-dev_1.1.0_amd64.deb

with the following commands:
`sudo dpkg -i libk4abt1.1-dev_1.1.0_amd64.deb `
`sudo dpkg -i libk4abt1.1_1.1.0_amd64.deb `

After that, I was able to run the k4abt_simple_3d_viewer from the command line (even if the computations were running on CPU) wit the following warning:

`2021-11-06 18:06:27.440808922 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign
2021-11-06 18:06:27.442896633 [W:onnxruntime:K4ABT_ORT_Environment, fallback_cpu_capability.h:140 GetCpuPreferedNodes] Force fallback to CPU execution for node: Shape__40`

After that, I supposed that the problem was the Cuda version so I decided to upgrade it to the latest one (the one that I actually have): Cuda 11.5.
(I am aware from other threads that the tested Cuda version with the body tracking is the Cuda11.1.1 but because of some Kernel problems I am not able to use that version and I **don't** want to downgrade my Kernel)

`k4aviewer` gives me absolutely no problems in running, all the cameras are perfectly working.
Here is the output as soon as I run from the terminal `k4abt_simple_3d_viewer`

`2021-11-06 18:12:43.229226675 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 GetCapability] CUDA kernel not found in registries for Op
2021-11-06 18:12:43.231439200 [W:onnxruntime:K4ABT_ORT_Environment, fallback_cpu_capability.h:140 GetCpuPreferedNodes] Force fallback to CPU execution for no
*** buffer overflow detected ***: terminated
Aborted (core dumped)`

No other logs are available.
So as you can see still the camera is not able to run with Cuda and moreover now does not even start anymore.
Any help will be very appreciated.

Niccolò

qm13, 2021-11-11:
We have not extensively tested on CUDA 11.5 and at this point only support CUDA 11.1.

pucciland95, 2021-11-14:
Hi @qm13 and thank you for your answer. What about the other part? I mean at least the CPU part should be working but it returns the previously said error. Any idea why?

qm13, 2021-11-15:
CPU mode still requires CUDA 11.1 dlls to be installed.

https://docs.microsoft.com/en-us/azure/kinect-dk/troubleshooting#required-dlls-for-onnx-runtime-execution-environments

cpkcpkcpk, 2021-11-22:
Same problem here - The Kinect Viewer is working fine, but the Body Tracking Viewer quits instantly on launch with no error messages. It also crashes when launching in CPU mode from command line.

PC:
AMD Ryzen 5900x
NVidia 3080
32GB RAM
Windows 10 21H1 19043.1348

Tried with both CUDA 11.1 and 11.5.

Any ideas on what's going on here? There are quite a few threads about this recently, so guessing it's a Windows or Nvidia update.

qm13, 2021-11-22:
@pucciland95 you do not mention what CPU you are using? Is it Intel or AMD?

pucciland95, 2021-11-22:
@qm13 I have added it to the initial question so everyone can see. Anyway it's an Intel(R) Core(TM) i7-1070H CPU 2.60 GHz

qm13, 2022-03-03:
Can you confirm that the skeleton ID for the first person in view switches to the second person when they come into view?

Another simple test is to use the k4abt_simple_3d_viewer tool and test whether the color of the first person changes when the second person comes in view.

qm13, 2022-03-03:
This is expected behavior. The message is a warning from ONNX Runtime telling you that there is no GPU implementation of a specific instruction and it is falling back to CPU. This typically happens with older GPUs.

ChangcongWang, 2022-03-07:
This issue can be resolved by #1696 and #1555. It seems like microsoft didn't make the adaption bewteen new vision driver ( Higher than 472 ) and SDK. I found the nvidia  driver you used is 495. What you need to just degrade your driver. I have the same probem few days ago, but I degrade driver to 470 version and it solve my problem. And I wish microsoft make the adaption bewteen new vision driver and SDK quickly.

fractalfantasy, 2022-03-21:
downgrading is not a sustainable solution as you can't downgrade newer graphics cards like the 3080ti to 470

cpkcpkcpk, 2022-03-25:
Seriously need a proper fix for this, that isn't downgrading to a 6 month old driver. It has been reported in multiple tickets for 6 months, and I need to keep my nvidia drivers up to date for lots of other software.

fkavache, 2022-04-07:
Hi, to those who had an issue on Windows, 
could you please take a look at this thread?

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1763#issuecomment-1091234984

fkavache, 2022-04-19:
@cpkcpkcpk

Please take a look at abovementioned thread, we need reports before official release.

12iqq, 2022-04-26:
Have the Updated Nvida driver: 512.15 (both studio and game)  but i guess i have to downgrade to make any of this work.... kinda sad it hasn't been fixed. Same issue as what's been described multiple times on this thread

fkavache, 2022-04-26:
@12iqq 

Could you please take a look at this thread?

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1763#issuecomment-1091234984

12iqq, 2022-04-27:
> 

That did fix the issue and it works now. Thanks!

closed 2022-05-25 by qm13

Omdiwan, 2023-07-28:
Hey, I am unable to downgrade my GPU driver to a lower version (536.67 -> 472.12) as it says it is not compatible with my windows version. Here are my specs
OS : win 11 home
processor : i7 12700h
ram : 16GB
GPU geforce 3060 rtx
Any fixes?