microsoft/Azure-Kinect-Sensor-SDK #1706: How to get undistort rgbd data from azure kinect dk？

zhangzscn, 2021-11-13:
Because some libraries do not support distortion parameters for reprojection，like pytorch3d。So i need the undistort rgbd data，just color image and depth image。

First，use function k4a_transformation_depth_image_to_color_camera() transforms the depth map from the viewpoint of the depth camera into the viewpoint of the color camera。

But after get the same viewpoint color image and depth image。**which camera distortion parameters to undistort color image and depth image？** Both use the color camera distortion parameters，or color image use the color camera distortion parameters and depth image use the depth camera distortion parameters。

**Which camera intrinsics parameters to transform depth image to pointcloud，the color camera or depth camera？**

Chang-Che-Kuei, 2021-11-15:
You have transformed the depth map into color space. So you should use color camera parameters to do the undistortion.

zhangzscn, 2021-11-16:
thank you！I have get the rgbd data form from azure kinect dk，by use opencv and color camera distortion parameters to undistort color image and depth image，then use color camera intrinsics parameters to transform depth image to pointcloud，the result pointcloud was as same as by use depth_image_to_point_cloud。

closed 2021-11-16 by zhangzscn