microsoft/Azure-Kinect-Sensor-SDK #1707: Failed to load shared object libdepthengine.so.2.0 with python sdk

zhangzscn, 2021-11-16:
After build the python sdk,get the k4a-0.0.2-py3-none-any.whl file,then install it to my project。
But when runing the examples of simple_viewer,get the error like this:

[2021-11-16 09:01:11.021] [error] [t=5427] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dynlib/dynlib_linux.c (82): dynlib_create(). Failed to load shared object libdepthengine.so.2.0 with error: libdepthengine.so.2.0: cannot open shared object file: No such file or directory
[2021-11-16 09:01:11.021] [error] [t=5427] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/deloader/deloader.cpp (75): deloader_init_once(). Failed to Load Depth Engine Plugin (depthengine). Depth functionality will not work
[2021-11-16 09:01:11.021] [error] [t=5427] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/deloader/deloader.cpp (76): deloader_init_once(). Make sure the depth engine plugin is in your loaders path
[2021-11-16 09:01:11.021] [error] [t=5427] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/deloader/deloader.cpp (113): deloader_depth_engine_create_and_initialize(). Failed to load depth engine plugin
[2021-11-16 09:01:11.021] [error] [t=5427] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 108.
[2021-11-16 09:01:11.021] [error] [t=5427] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (160): deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
[2021-11-16 09:01:11.021] [error] [t=5427] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (194): depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread()
[2021-11-16 09:01:11.021] [error] [t=5419] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (552): dewrapper_start(). Depth Engine thread failed to start
[2021-11-16 09:01:11.021] [error] [t=5419] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (398): dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
[2021-11-16 09:01:11.027] [error] [t=5419] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[2021-11-16 09:01:11.028] [error] [t=5419] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[2021-11-16 09:01:11.031] [error] [t=5419] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (895): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()
[2021-11-16 09:01:11.031] [error] [t=5419] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (112): queue_pop(). Queue "Queue_capture" was popped in a disabled state.
[2021-11-16 09:01:11.031] [error] [t=5419] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (269): capturesync_get_capture(device->capturesync, capture_handle, timeout_in_ms) returned failure in k4a_device_get_capture()

zhangzscn, 2021-11-16:
just move libdepthengine.so.2.0 to /usr/lib/x86_64-linux-gnu solve the problem

closed 2021-11-16 by zhangzscn

rajkundu, 2024-05-16:
I can confirm - running `sudo ln -s /usr/lib/x86_64-linux-gnu/libk4a1.4/libdepthengine.so.2.0 /usr/lib/x86_64-linux-gnu/libdepthengine.so.2.0` worked for me!