microsoft/Azure-Kinect-Sensor-SDK #1712: Enumeration errors on Ubuntu 18.04 NVidia Jetson Nanos

michael-createc, 2021-11-29:
On NVidia Jetson Nanos running Ubuntu 18.04, if the Azure Kinect is already plugged in to the USB port and externally powered before the Nano switches on, then the device will almost never enumerate (maybe it'll enumerate 1 in every 15-20 attempts). We have encountered this issue with at least 10 different Azure Kinects combined with at least 15 different Jetson Nanos. This issue has been mentioned in some form or another in issues #1598, #1260 and #375, but all have been closed without resolution...

Unplugging the Kinect and from either external power or the USB port, and then plugging it back in, solves the issue and makes everything work fine. However, we are dealing with a distributed system involving many Kinects and Nanos here, and to have to go and manually unplug and replug every one each time you turn on the system is not plausible

**Logs**
As mentioned in the other issues, `dmesg` gives this type of thing on repeat:
```
[ 1188.573468] usb 2-1.4-port2: attempt power cycle
[ 1188.979200] usb 2-1.4.2: Device not responding to setup address.
[ 1189.203254] usb 2-1.4.2: Device not responding to setup address.
[ 1189.414953] usb 2-1.4.2: device not accepting address 46, error -71
[ 1189.759218] usb 2-1.4.2: Device not responding to setup address.
[ 1189.983188] usb 2-1.4.2: Device not responding to setup address.
[ 1190.194930] usb 2-1.4.2: device not accepting address 47, error -71
[ 1190.201405] usb 2-1.4-port2: unable to enumerate USB device
[ 1190.691190] usb 2-1.4.2: Device not responding to setup address.
[ 1190.915188] usb 2-1.4.2: Device not responding to setup address.
[ 1191.126900] usb 2-1.4.2: device not accepting address 48, error -71
[ 1191.471176] usb 2-1.4.2: Device not responding to setup address.
[ 1191.695164] usb 2-1.4.2: Device not responding to setup address.
[ 1191.908318] usb 2-1.4.2: device not accepting address 49, error -71
```

And the firmware tool with info level logging:
```
AzureKinectFirmwareTool -l
 == Azure Kinect DK Firmware Tool == 
[2021-11-29 18:10:14.864] [critical] [t=17] ../src/usbcommand/usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2021-11-29 18:10:14.864] [error] [t=17] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2021-11-29 18:10:14.942] [info] [t=17] ../src/usbcommand/usbcommand.c (343): find_libusb_device(). Container ID found: {fb989d7f-f523-4fb0-c68e-761838afef6e} 
[2021-11-29 18:10:14.943] [info] [t=17] ../src/usbcommand/usbcommand.c (224): populate_serialnumber(). Serial Number found 000746401612
Found 1 connected devices:
1: Device "000746401612"
```

Once again, I will say that all of the above issues go away when you unplug and replug the Kinect, but we cannot afford to do this for every Kinect, every time.

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04 on a Jetson Nano, L4T version 32.2.0.
 - SDK Version:  Installed from source off latest develop branch
 - Firmware version:  
RGB camera: 1.6.102
Depth camera: 1.6.75
Depth config: 6109.7

qm13, 2021-12-07:
@michael-createc you need to upgrade to the latest firmware [here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/firmware).

alexhyy, 2022-04-21:
It was fixed by [this firmware](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/firmware) already.

michael-createc, 2022-04-21:
Can confirm that firmware update fixed the issue thanks!

closed 2022-04-21 by michael-createc