microsoft/Azure-Kinect-Sensor-SDK #1713: Azure Kinect extern Sync 1536p / 15FPS capture flicker

KR-ESWord, 2021-12-01:
Hello!
I'm shooting using the extetnal Sync mode of multiple Azure Kinects.
Using the setting as below.

Master Mode Azure Kinect Setting
- NFOV Unbinned
- BGRA
- 1536p
- 15FPS

Subordinate Mode Azure Kinect Setting
- NFOV Unbinned
- BGRA
- 1536p
- 15FPS

When shooting with Azure Kinect with these settings, the image flickered in Subordinate Mode Azure Kinects.

(Left : Subordinate Azure Kinect / Right : Master Azure Kinect)
[image]
[image]

If the Master Azure Kinect is 1536p at 15FPS, the resolution of the Subordinate Azure Kinect is not 1536p, but it seems that the Master 
 Azure Kinect is not sending the signal normally because of the flickering phenomenon.

How can I fix this?

Thanks!

qm13, 2021-12-07:
@KR-ESWord from the screen shots it does not look like you have programmed in a delay for each of the subordinates. I would expect to see flashing if the delay is 0 for all the subordinates as they would all be taking a depth image at the same time.

KR-ESWord, 2021-12-07:
> I'm gave various Delay off master to 100us, 300us, 1000us, ... , 10000us, 30000us, ... , 50000us.
but, they were still in the same condition.

qm13, 2021-12-07:
@KR-ESWord did you assign different delays to each of the subordinates? Please [see](https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync#avoiding-interference-between-multiple-depth-cameras).

KR-ESWord, 2021-12-09:
> I gave to Subordinate Azure Kinect to 160us delay off master. but still same condition.
this my test case. I use one Master Azure Kinect and one Subordinate Azure Kinect.
1. Master : 0us Depth delay / Subordinate : 160us Delay off master
2. Master : 160us Depth delay / Subordinate : 160us Delay off master
3. Master : 106us Depth delay / Subordinate : 0us Delay off master

qm13, 2021-12-09:
@KR-ESWord can you verify if this problem is present with different RGB modes e.g.1080p RGB at 15 and 30 fps, and with NFOV Binned? Just use a 160us delay on the subordinate device.
Also please conductivity of your sync cable.

KR-ESWord, 2021-12-10:
> @KR-ESWord can you verify if this problem is present with different RGB modes e.g.1080p RGB at 15 and 30 fps, and with NFOV Binned? Just use a 160us delay on the subordinate device. Also please conductivity of your sync cable.

This phenomenon only occurs in Color Resolution : 1536p & Frame Rate : 15FPS.
It was normal to choose a different frame rate at a different resolution.

KR-ESWord, 2022-01-03:
@qm13 Has it been updated?

KR-ESWord, 2022-04-13:
Has it been updated?

KR-ESWord, 2022-05-24:
@dasparli Hi! Has it been updated?

dasparli, 2022-05-26:
I suspect the flickering of the RGB frames on the subordinate camera is caused by dropped frames rather than a problem with frame synchronization. Do you have both cameras connected to the same USB hub? If so, please try connecting them to completely separate USB hubs.

KR-ESWord, 2022-05-27:
@dasparli Thanks for replying.
But, Even after connecting to a different PC for each Azure Kinect and configuring it as a Star Chain, the flickering of RGB frames on the subordinate camera still occurs when set to 15 FPS at 1536 resolution.

dasparli, 2022-06-15:
Perhaps there is a problem with your camera. Do you have other cameras you can try? Have you tried swapping the master/subordinate roles between the two physical units?

KR-ESWord, 2022-06-16:
@dasparli I tried changing the Master/Subordinate connection by 3 units with about 10 Azure Kinects, but still the flickering phenomenon is found when starting at 15 fps at resolution 1536p.

KR-ESWord, 2023-04-14:
@dasparli Has it been updated?

realElMartino, 2024-05-29:
Any solution? I have the same issue with the same settings. Other settings work.