microsoft/Azure-Kinect-Sensor-SDK #1719: Azure Kinect DK Getting: 'Pixels not aligned to stride of each line'

Alessandro-Minerba, 2021-12-27:
0

I'm using the Kinect DK and I'm trying to save the camera Capture (a single frame with Color, Depth, IR images). I've already realised that to save a Capture object I have to save the 3 single images (cited above) and than to reconstruct the Capture from the saved files. I've successfully saved the Color image, but I'm having some issues with Depth and IR, because of their format. The problem occurs when I try to create the var a = calibration.CreateTransformation() which I think is necessary to use its function a.DepthImageToColorCamera(capture.Depth, depthImage) to return an image with the specified dimensions of depthImage object. The system returns a <Microsoft.Azure.Kinect.Sensor.AzureKinectException: 'Pixels not aligned to stride of each line'> exception. Have someone had the same issue? Thanks.

P.S. I'll give you the link to microsoft documentation https://docs.microsoft.com/it-it/azure/kinect-dk/use-image-transformation

string pathDepth = @"C:\Users\MINALE\Desktop\Kinect_app\depth.json";

            if (!File.Exists(pathDepth))
            {
                // Create a file to write to.
                using (StreamWriter sw = File.CreateText(pathDepth))
                {
                    JsonSerializer serializer = new JsonSerializer { Formatting = Formatting.Indented };

                    Image depthImage = new Image(ImageFormat.Depth16, 512, 512);

                    try
                    {
                        var a = calibration.CreateTransformation();
                        a.DepthImageToColorCamera(capture.Depth, depthImage);
                    }
                    catch (Exception ex)
                    {

                    }

                    serializer.Serialize(sw, depthImage.GetPixels<BGRA>().Span.ToArray());

                }

            }

closed 2022-01-04 by Alessandro-Minerba

asergaz, 2022-01-07:
Relates with https://stackoverflow.com/questions/70461998/azure-kinect-dk-getting-pixels-not-aligned-to-stride-of-each-line .

asergaz, 2022-01-07:
Looks like issue was solved [here ](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1208) but the solution was not shared.

asergaz, 2022-01-11:
See how it is done here: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/tools/k4arecorder